eclipse-embed-cdt/eclipse-plugins #2: Where can i configure and manage a gdbinit script for DGB SEGGER J-Link Debugging

proteusmk, 2015-09-19:
Hello ilg-ul,

my development environment ist eclipse mars latest cdt and gnuarmeclipse plugin and all things are working pretty. Now i would like to add a gdbinit file like old settings from pure gdb sessions. In my old environment i add a .gdbinit with following settings:

define hook-step
mon cortex_m3 maskisr on
end
define hookpost-step
mon cortex_m3 maskisr off
end

In the source file ilg.gnuarmeclipse.debug.gdbjtag.jlink/src/ilg/gnuarmeclipse/debug/gdbjtag/jlink/Configuration.java at line 276 is a comment that these settings are set in the LaunchSequence.  Now i thing that the arm-none-eabi-gdb --interpreter=mi2 --nx process on linux is a client process, so i look in the main documentation from JLinkGdbServer for information to handle the maskisr on and of function.

My question: can you tell me some examples how i can enable maskisr?

Another cool topic:
Now i have test some settings to flash a binary into the LPC1768 chip with J-Link, here are my results:

You can  set up a External Tools RUN configuration in eclipse to start the JLinkExe program directly  with parameter to flash a bin file directly in ROM. So you don't need to start a debug session for the same result and it is very fast.

[image]

On Location: /opt/SEGGER/JLink/JLinkExe
On Working Directory: ${workspace_loc:/LPC17xx_Blinky}
On Arguments: -device LPC1768 -speed 1000khz -CommanderScript ${workspace_loc:/LPC17xx_Blinky}/openocd/lpc17xx_flash.jlink

File lpc17xx_flash.jlink:
si
speed 4000
r
h
loadbin /projekte/eclipse/workspace450/LPC17xx_Blinky/Debug/LPC17xx_Blinky.bin,0x0
verifybin /projekte/eclipse/workspace450/LPC17xx_Blinky/Debug/LPC17xx_Blinky.bin,0x0
q

The same in a bash script jlink_lpc17xx_flash.sh:
#!/bin/bash -
#===============================================================================
#
#          FILE: jlink_lpc17xx_flash.sh
#
#         USAGE: ./jlink_lpc17xx_flash.sh
#
#   DESCRIPTION: Start JlinkExe with jlink CommanderScript to flash the program in ROM
#
#       OPTIONS: $1 is the workspace project
#  REQUIREMENTS: ---
#          BUGS: ---
#         NOTES: ---
#        AUTHOR: Michael Kruschinsky (), mk@mysmk.de
#  ORGANIZATION: mySMK
#       CREATED: 19.09.2015 00:23:24 CEST
#      REVISION:  ---
#===============================================================================

set -o nounset                              # Treat unset variables as an error

#/opt/SEGGER/JLink/JLinkExe -device LPC1768 -speed 1000khz -CommanderScript /projekte/eclipse/workspace450/LPC17xx_Blinky/openocd/lpc17xx_flash.jlink
/opt/SEGGER/JLink/JLinkExe -device LPC1768 -speed 1000khz -CommanderScript $1/openocd/lpc17xx_flash.jlink

regard
Michael

ilg-ul, 2015-09-19:
> can you tell me some examples how i can enable maskisr?

unfortunately not, perhaps a better place to ask J-Link related questions is SEGGER support.

> Another cool topic: ...

I'm not sure I get the point. did you discover a new way to program devices? ok, very nice.

ilg-ul, 2015-09-21:
I checked the source code and indeed, the GDB init scripts are deliberately ignored.

in case your question was not specific to maskisr, but more generic, related to passing commands to GDB, I updated [J-Link](http://gnuarmeclipse.github.io/debug/jlink/#no-gdb-initialisation-files-loaded) page to explain how to do it (although it should have been obvious).

closed 2015-10-16 by ilg-ul