eclipse-embed-cdt/eclipse-plugins #5: Missing Tool Settings Tab

abencomo, 2015-09-26:
I've followed all the installation steps and I'm able to run both tutorials (Blinky ARM test & Hello ARM test) without problems.  However, when I import a project into Eclipse via Import > "Existing Code as Makefile Project", then the Properties - Setting dialog is missing the "Tool Settings" tab and others tabs (see attachments). To reproduce, just reimport the Hello ARM test project.

Eclipse IDE for C/C++ Version: Mars Release (4.5.0) Build id: 20150621-1200
gnuarmeclipse-qemu-osx-2.3.50-201508041609-dev
gcc-arm-none-eabi-4_9-2015q3
OSX 10.10.4

[image]

[image]

[image]

ilg-ul, 2015-09-26:
>  Import > "Existing Code as Makefile Project" ... then the Properties - Setting dialog is missing the "Tool Settings" tab

what Tool Settings you would like to set for a **Makefile project**?

the only setting available for unmanaged projects is the toolchain, used by the indexer, and this settings, as seen before, is available.

the rest of the settings are available only for managed C/C++ projects.

abencomo, 2015-09-26:
I'm not sure why you labeled your reply as a question when it seems that you're actually proving a answer to it already (i.e. "the only setting available for unmanaged projects is the toolchain"). 

Anyhow, I guess it is just confusing that the same hello project can have different settings based on how it was added into the workspace.  So, if you create the hello project via File > New > C Project, then it is managed.  However, if you reimport this originally managed project back to the workspace via Import > Existing Code as Makefile Project, then it becomes unmanaged.  Is that what you meant with your Question/Answer?

closed 2015-09-27 by abencomo

abencomo, 2015-09-27:
I figured out how to bring the Tool Settings Tab back.  This was not obvious to me so I'm documenting it here in case someone else has the same question.  In the project properties > C/C++ Build > tab "Builder Settings", just check the box for the option "Generate Makefiles automatically" and that will bring back the Tool Settings Tab as shown below.

BEFORE:
[image]
[image]

AFTER:
[image]
[image]

ilg-ul, 2015-09-27:
Your conclusion about managed and unmanaged projects is mostly correct, just that generally not all unmanaged projects can be converted to managed with a single click. Most complex makefiles have hand-written rules that cannot be automatically generated by Eclipse.

The mistake in your approach was to use **Import** -> **Existing Code as Makefile Project** instead of **Import** -> **General** -> **Existing Projects into Workspace**. Always use _Existing Projects_ for projects created by Eclipse, regardless of their type  (i.e. if you have a `.project` file, possibly a `.cproject` file for C/C++ projects, Eclipse will know its type and how to import it). Use _Existing Code as Makefile Project_ only for projects that were created manually, and have a Makefile but do not have a `.project` file.