eclipse-embed-cdt/eclipse-plugins #6: #error "<dirent.h> not supported"

abencomo, 2015-09-28:
I downloaded all the components required by the GNU ARM Eclipse documentation. However, when I try to build my project on OSX Yosemite (10.10.4); I get this  #error "<dirent.h> not supported".  Do I need to install something else.  Thank you in advance!

-----------
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
 
Building file: ../psp/fsw/shared/cfe_psp_eeprom.c
Invoking: Cross ARM C Compiler
arm-none-eabi-gcc -mcpu=cortex-a7 -mthumb -O2  -g -I../include -I/Users/abencomo/Downloads/cFE-6.4.2b/psp/fsw/inc -I/Users/abencomo/Downloads/cFE-6.4.2b/osal/src/os/inc -I/Users/abencomo/Downloads/cFE-6.4.2b/osal/build/inc -std=gnu11 -MMD -MP -MF"psp/fsw/shared/cfe_psp_eeprom.d" -MT"psp/fsw/shared/cfe_psp_eeprom.o" -c -o "psp/fsw/shared/cfe_psp_eeprom.o" "../psp/fsw/shared/cfe_psp_eeprom.c"
In file included from /usr/local/gcc-arm-none-eabi-4_9-2015q3/arm-none-eabi/include/dirent.h:6:0,
                 from ../psp/fsw/shared/cfe_psp_eeprom.c:28:
/usr/local/gcc-arm-none-eabi-4_9-2015q3/arm-none-eabi/include/sys/dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"
  ^

abencomo, 2015-09-28:
I noticed that there are two(2) dirent.h files inside gcc-arm-none-eabi-4_9-2015q3.  However, eclipse seems to pick up only the one under the /sys directory instead of the one above.

This one:
/usr/local/gcc-arm-none-eabi-4_9-2015q3/arm-none-eabi/include/sys/dirent.h

Instead of this one:
/usr/local/gcc-arm-none-eabi-4_9-2015q3/arm-none-eabi/include/dirent.h

abencomo, 2015-09-28:
The sys/dirent.h has header saying: 

/* ...this file is overridden by dirent.h in the libc/sys/.../sys directory. ..*/

Since my target platform is Raspberry Pi 2 armv7l Cortex-A7, then I think I need to bring the dirent.h from the RPi2 and copy it onto the mac running.

ilg-ul, 2015-09-28:
as the name implies, _GNU Tools for ARM Embedded Processors_ is intended for bare metal applications. if you need to build applications for ARM GNU/Linux, you probably need to install a Linaro toolchain.

http://gnuarmeclipse.github.io/toolchain/install/

anyway, the issue is not related to GNU ARM Eclipse, but to the toolchain use.

abencomo, 2015-09-28:
But I already installed and setup the toolchain as that documentation indicated:

[bencomoa:local/gcc-arm-none-eabi-4_9-2015q3/bin] abencomo% arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

abencomo, 2015-09-28:
Or, are you trying to say is that I need to switch the toolchain to one of the options below?

[image]

ilg-ul, 2015-09-28:
For Raspberry Pi 2 you can write either Linux or bare-metal applications, little or big endian. Select the proper Linaro toolchain (probably arm-linux-gnueabihf-gcc).

abencomo, 2015-09-28:
I think I know what you meant before.  I need to install the Linaro toolchain from the link below.  If that is the case, do you know which one will be the best for the Raspberry Pi 2?  Thanks!

http://www.linaro.org/downloads/

abencomo, 2015-09-28:
As you suggested, I switched to arm-linux-gnueabihf-gcc, but I now get this error:

/bin/sh: armeb-linux-gnueabihf-gcc: command not found
make: *** [psp/fsw/shared/cfe_psp_eeprom.o] Error 127

[image]

ilg-ul, 2015-09-28:
Please read thoroughly the documentation.

closed 2015-10-16 by ilg-ul