eclipse-embed-cdt/eclipse-plugins #7: OpenOCD silent installations on windows 7

rgkandur, 2015-09-29:
I am trying to install https://github.com/gnuarmeclipse/openocd/releases/download/gae-0.9.0-20150519/gnuarmeclipse-openocd-win64-0.9.0-201505190955-setup.exe from command line. I am successfully able to install silently with /S option but failed to install to different folder than default.  The silent installations always installing openocd at "C:\Program Files\GNU ARM Eclipse\OpenOCD".  Kindly let me know how install to different folder than default one.

Following are the commands i used to install openocd at "L:\openocd":

C:\>"C:\Users\rkandur\Downloads\gnuarmeclipse-openocd-win64-0.9.0-201505190955-setup.exe" /S /D=l:\openocd

ilg-ul, 2015-09-29:
This tracker is only for issues related to the Eclipse plug-ins. Please address OpenOCD issues to the OpenOCD tracker.

closed 2015-09-29 by ilg-ul

bitsmack, 2015-10-30:
Please disregard this post.  It is just a test to see if I can comment on closed issues.