eclipse-embed-cdt/eclipse-plugins #8: Disconnecting after succesfull run

bremenpl, 2015-10-08:
Hello there,
I am happy to see the project migrated to github, the new site is great as well :)!

I would like to bring up front the case with automatic disconnect feature after successful run again. I am working heavilly with you plugin now and I notice that this feature would be really usefull and would improve work a lot (at least in my opinion). I was wondering either it would be possible for you to add a checkbox in the debug configuration maybe so that after checking it, after a run configuration is used, JLINK would automatically disconnect from target?

The plugin is great at this point, so thank you anyways :).

ilg-ul, 2015-10-08:
Thank you for the suggestion.

You mean to disconnect from the target, and leave it running? I thought that, if the target is running, it will remain running, and this should already be implemented. I don't know what should happen if the target is halted, I'm not sure it should continue or not.

bremenpl, 2015-10-08:
At the moment when you use run configuration (not debug) the target is being programmed but jlink remains connected. If I want to run another run session I need to disconnect first, it doesnt disconnect automatically- thats what I mean.

ilg-ul, 2015-10-08:
Ah, ok, that's a bit different, I'll investigate.

bremenpl, 2015-10-08:
Thank you :). I really think this is an usefull feature.

bremenpl, 2015-10-08:
This is a screenshot of what I mean:

[image]

I have to disconnect manually at the bottom and also adress 0x0 tab is being opened (not always but mostly yes).

ilg-ul, 2015-10-27:
I tried to find a solution for this, but things are quite complicated, I'm afraid I have no time now for it.

bremenpl, 2015-10-28:
I understand. Too bad but thank you for the efford :).

bremenpl, 2015-10-28:
I have a question though- have you thought maybe about looking at the 
problem from a bit different angle? Wouldnt it be esier at every point, if 
run configuration would not connect at all? What if run configuration would 
simply program the target and reset it? Couldnt run configuration use a 
flasher script to program the target? All problems regarding disconnecting 
and killing processes would dissapear. And the most importantly, the 
functionality of a run config would stay the same.

ilg-ul, 2015-10-28:
implementing the Run configuration was just a quick and dirty kludge, it uses exactly the same infrastructure as the Debug configuration, just that it disables breakpoints, so it cannot stop.

of course better solutions are available, but they definitely require more time to implement.

if you provide functional pull requests, I can consider integrating them.

bremenpl, 2015-10-28:
I was considering it really, but all Ive ever written in java is hello 
world :p.

ilg-ul, 2015-10-28:
Java itself is not a big problem, Eclipse helps a lot when developing Java code; much more difficult is to understand the Eclipse framework, even more complicated is CDT, and much more difficult is DSF, the debug framework.

bremenpl, 2015-10-28:
In that case its even more not me... Anyways, I really apreciate you help 
and feedback. Im gratefull for the way it works now :).
For the time being, before you find the time ill just add mysalge an 
external tool script.