eclipse-embed-cdt/eclipse-plugins #9: J-Link and LPCXpresso (Eclipse-based IDE from NXP) debug issue

pozzugno, 2015-10-09:
I'm using LPCXpresso 7.9.2_493 (that is an Eclipse-based IDE released from NXP, I think it is based on Eclipse 4.4.2). I'd like to use J-Link debugger/probe in LPCXpresso.

Segger [suggests](https://www.segger.com/nxp-lpcxpresso.html) to use GNU ARM Eclipse plug-ins as I did just now.

**First problem**: I can't see Devices tab in the Settings window of project Properties. It seems this is not mandatory, so I didn't care about it.

I created a new Debug Configuration and set the cross_prefix substring (as indicated by Segger in this [link](https://www.segger.com/nxp-lpcxpresso.html)).

**Second problem**: when I launch the debug session, the error **Error while launching command: arm-none-eabi-gdb --version** appears. I think gdb client can't be found, so I changed the executable from ${cross_prefix}gdb${cross_suffix} to C:\nxp\LPCXpresso_7.9.2_493\lpcxpresso\tools\bin\arm-none-eabi-gdb.exe.

Now it seems to work.

ilg-ul, 2015-10-09:
> I can't see Devices tab in the Settings window of project Properties

the Devices tab is part of the Packs plug-ins.

> Error while launching command: arm-none-eabi-gdb --version

as you noticed, that's because the toolchain path is not set. the GNU ARM Eclipse plug-ins use an elaborated mechanism to set the toolchain path, that does not match the LPCXpresso mechanism.

I'm not sure if there is anything to do to make things easier.

pozzugno, 2015-10-09:
> the Devices tab is part of the Packs plug-ins.
I installed GNU ARM Packs
[image]

ilg-ul, 2015-10-09:
Ah, sorry, my reply was not complete; I just checked, and it is correct that the Devices tab is part of the Packs plug-ins, but it is visible only for projects managed by the GNU ARM Eclipse build plug-in, so it is not possible to assign devices/boards to projects created by other plug-ins, because they lack the mechanisms to handle such additional data.

ilg-ul, 2015-10-16:
I added two separate issues, to make the Devices tab visible and to not use macros for projects not created with GNU ARM Eclipse, so I think this ticket can be closed now.

closed 2015-10-16 by ilg-ul

ilg-ul, 2015-10-26:
I published a beta version on updates-test, that makes the Devices tab visible for all projects, so it should be visible for LPCXpresso projects too.

Could you test it and confirm if it works properly for you?