eclipse-embed-cdt/eclipse-plugins #10: Semihosting check when not running under debugger

egoltzman, 2015-10-12:
The FAQ say:
"
Update: M3/M4 projects generated with version 2.2.1-201404120702 or later check when not running under debugger and skip the trace call, so it is possible to run the application standalone.
"

But how should it set? I have 2 projects using the trace_printf with semihosting, one is working fine without debugger and the other stuck but I can't find the different, where is this issue being taking care of?

Thanks

ilg-ul, 2015-10-12:
I guess you are referring to the new `HardFault_Handler()` In [exception_handlers.c](https://github.com/gnuarmeclipse/plug-ins/blob/develop/ilg.gnuarmeclipse.templates.core/templates/common/system/src/cortexm/exception_handlers.c).

egoltzman, 2015-10-12:
That is exactly what I looked for !!! Fast and efficient response - Thanks 
Just for the record, the cause for the problem in my case was two definitions of the HardFault_Handler, the second one was stronger then the  exception_handlers.c weak one...

closed 2015-10-12 by egoltzman

ilg-ul, 2015-10-12:
BTW, why did your have to define the HardFault_Handler? Do you need to do something more? If so, can you suggest improvements to the existing handler?