eclipse-embed-cdt/eclipse-plugins #12: -c echo "Started by GNU ARM Eclipse" kills openOCD

M3gaFr3ak, 2015-10-15:
When I use "StartOpenOCD locally", debugging doesn't work because openOCD crashes because the plugin attatches "-c echo "Started by GNU ARM Eclipse"" to the arguments. Launching OOCD w/o this argument, it works flawlessly

ilg-ul, 2015-10-15:
> ... Launching OOCD w/o this argument, it works flawlessly

how did you test this? what openocd version are you using?

M3gaFr3ak, 2015-10-15:
Version 0.9.0

I tested it by trying in eclipse, then trying by launching oOCD in the terminal

ilg-ul, 2015-10-15:
please retest with the latest GNU ARM Eclipse OpenOCD in the terminal and perform two runs, with and without the extra command, and pass me the output. be sure you use exactly the same command line as issued by Eclipse (you can get the command line with right click on the openocd process).

M3gaFr3ak, 2015-10-15:
what do you mean with latest GNU ARM Eclipse oOCD? does it provide one in the package?

ilg-ul, 2015-10-15:
http://gnuarmeclipse.github.io/openocd/

M3gaFr3ak, 2015-10-15:
using this makes it work, thanks!

closed 2015-10-15 by M3gaFr3ak

ilg-ul, 2015-10-21:
What [PSA] stands for, and how does this addition help in this issue?

M3gaFr3ak, 2015-10-21:
Public service announcement.
I feel like this issue might occur frequently due to the difference between oOCD and this open ocd modification

ilg-ul, 2015-10-21:
>  this open ocd modification

currently there are no major changes, it just fixes some small bugs related to automatically locating the scripts folder.

M3gaFr3ak, 2015-10-21:
yes, but apparently the "-c echo ..." stuff doesn't work with "normal" open ocd (which was the core of my issue)

ilg-ul, 2015-10-21:
did you perform the test I suggested, i.e. two runs of the normal openocd, in a terminal, with and without the `-c echo xxx`, using exactly the command line used by the plug-ins?

M3gaFr3ak, 2015-10-21:
yes, tried it. error in commandline:

"none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
in procedure 'echo'"

w/o -c echo "..." it works properly

19:00 Uhr:

ilg-ul, 2015-10-21:
please copy me the exact command line used and the output. be sure you use quotes with `-c 'echo xxx'`

M3gaFr3ak, 2015-10-21:
I call
openocd -f "C:\Program Files\GNU ARM
Eclipse\OpenOCD\0.9.0-201505190955\scripts\board\stm32f4discovery.cfg"
-c echo "cake""

which gives me the error. If I add parentheses to the echo stuff, it has
the same error, but adds "invalid command name "echo""

eclipse also seems to not use parentheses (in the echo stuff)

20:32 Uhr:

mrtumnus, 2015-10-21:
You need a set of double parentheses around your command argument:
openocd -f "C:\Program Files\GNU
ARM Eclipse\OpenOCD\0.9.0-201505190955\scripts\board\stm32f4discovery.cfg" -c
"echo cake"

M3gaFr3ak, 2015-10-21:
that works. but eclipse calls it w/o parentheses

command eclipse calls:
Error in services launch sequence
Launching command [C:\Program Files\openocd-0.9.0\bin\bin\openocd.exe -c
gdb_port 3333 -c telnet_port 4444 -f C:\Program
Files\openocd-0.9.0\bin\scripts\board\stm32f4discovery.cfg -c echo "Started
by GNU ARM Eclipse"] failed.
Launching command [C:\Program Files\openocd-0.9.0\bin\bin\openocd.exe -c
gdb_port 3333 -c telnet_port 4444 -f C:\Program
Files\openocd-0.9.0\bin\scripts\board\stm32f4discovery.cfg -c echo "Started
by GNU ARM Eclipse"] failed.
Cannot run program "C:\Program Files\openocd-0.9.0\bin\bin\openocd.exe":
Launching failed

um 20:43 Uhr:

ilg-ul, 2015-10-21:
> a set of double parentheses

they are called quotes, or double quotes.

M3gaFr3ak, 2015-10-21:
ah sorry, not a native speaker^^

20:48 Uhr:

mrtumnus, 2015-10-21:
:facepalm:

Sorry, had a brain fart there.

M3gaFr3ak, 2015-10-21:
but yeah, eclipse doesnt have quotes in the command (it has them, but they
dont encapsulate the echo)

um 20:51 Uhr:

ilg-ul, 2015-10-21:
> eclipse calls it w/o parentheses

nope. Eclipse calls the process with an array of strings for argv[], the quotes are required only for the shell (cmd.exe for Windows users), to split the line intro separate strings.

there must be another problem.

ilg-ul, 2015-10-22:
did you identify what is wrong with your initial configuration? did you use Freddie Chopin binary, or you built it yourself?

M3gaFr3ak, 2015-10-22:
I used the prebuilt binaries. Didn't identify anything really, no... But it
works w/ the modified openocd curiously ôO

15:27 Uhr:

ilg-ul, 2015-10-22:
> I used the prebuilt binaries. 

hmmm... what prebuilt binaries? don't you remember where you got them from? exactly what version it was?

ilg-ul, 2015-10-22:
... and please **do not reply by email**, use the web interface, especially when you are quoting the original messages.

M3gaFr3ak, 2015-10-22:
built by freddie chopin.

0.9.0

ilg-ul, 2015-10-24:
we did some tests and Freddie's v9 and v10.dev  binaries, on both 32/64-bits, seem ok.

I suspect a local configuration problem on your machine.

lberezy, 2015-11-12:
I'd like to chime in and say I am having a very similar/same issue.
I've just come across GNU ARM Eclipse today and have set it up according to the documentation and have used this plugin. Eclipse tries to execute the following `/Applications/GNU\ ARM\ Eclipse/OpenOCD/0.10.0-201510281129-dev/bin/openocd -c gdb_port 3333 -c telnet_port 4444 -f board/stm32f4discovery.cfg -c echo "Started by GNU ARM Eclipse"`, the same as the problem already discussed here with the same error.
This is what happens when I run the above in a terminal:
```
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00141-g09aeb96 (2015-10-28-13:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
3333
4444
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
in procedure 'echo'
```

Modifying the command to `/Applications/GNU\ ARM\ Eclipse/OpenOCD/0.10.0-201510281129-dev/bin/openocd -c gdb_port 3333 -c telnet_port 4444 -f board/stm32f4discovery.cfg` (dropping the echo) produces what I believe to be workable results complete with blinking lights from the STLink portion of my Discovery board (STM32F429I) (I've yet to get my board flashed and running from inside Eclipse, still setting up the toolchain, so I'm not sure of what's meant to happen):
```
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00141-g09aeb96 (2015-10-28-13:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
3333
4444
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v23 API v2 SWIM v0 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.865149
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
```

ilg-ul, 2015-11-12:
when starting openocd from shell, quote the entire echo command:

```
-c 'echo "Started by GNU ARM Eclipse"'
```

lberezy, 2015-11-12:
Yes, that removes the error, but the problem still remains. It's this plugin that appends `-c echo "Started by GNU ARM Eclipse"` and I have no way that I can see of telling it not to. Thus, the plugin cannot launch OOCD - I thought this was pretty obvious from the text of the message that's trying to be `echo`'d.

ilg-ul, 2015-11-12:
> the plugin cannot launch OOCD

sorry, on my machine it starts just fine:

```
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00141-g09aeb96 (2015-10-28-13:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Started by GNU ARM Eclipse
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v14 API v2 SWIM v0 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.911943
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : stm32f4x errata detected - fixing incorrect MCU_IDCODE
Info : device id = 0x10006413
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002b8 msp: 0x20020000
semihosting is enabled
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002b8 msp: 0x20020000, semihosting
target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000042 msp: 0x20020000, semihosting
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002b8 msp: 0x20020000, semihosting
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002b8 msp: 0x20020000, semihosting
===== arm v7m registers
(0) r0 (/32): 0x00000000
```

if you think there is a problem with the plug-in, you need to provide enough information to reproduce the problem.

> It's this plugin that appends -c echo "Started by GNU ARM Eclipse" and I have no way that I can see of telling it not to

why would you want to tell it not to?

passing commands on the command line with `-c` is perfectly legal, the `echo` command is implemented in the standard OpenOCD, so it is not a problem to be used; I don't understand all the fuss around this command. when you start OpenOCD by hand you don't need to use it, because the plug-in does not wait for it. if, for any reasons, you want to enter it by hand, just quote the entire content.

I suggest you do a clean test, remove all other OpenOCD except the last one, install a clean Eclipse, create a new workspace, generate a new blinky project with the F4 template and try again.

if it still does not work, create a new ticket, provide all details, provide the openocd logs, the gdb traces, etc, and be sure you do not blame the echo command again, since it has nothing to do with your problem.

jacobjennings, 2016-04-05:
Using Kinetis Design Studio 3.2.0 (eclipse) on Mac OSX 10.11.3 (15D21) with openocd debug eclipse plugin version:

  GNU ARM C/C++ OpenOCD Debugging	3.2.1.201602101653 ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group	Liviu Ionescu

and openocd version:

Open On-Chip Debugger 0.10.0-dev-00189-g554313b (2016-01-06-13:33)

I see the exact same issue. 

`
$ openocd -c gdb_port 2331 -c telnet_port 2333 -f ~/openocd.cfg -c echo "Started by GNU ARM Eclipse"
Open On-Chip Debugger 0.10.0-dev-00189-g554313b (2016-01-06-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
3333
4444
adapter speed: 8000 kHz
Info : add flash_bank kinetis kl17.flash
adapter speed: 1000 kHz
none separate
cortex_m reset_config sysresetreq
Info : add flash_bank kinetis pflash.0
in procedure 'echo'
`

If I add single quotes around the echo,

`
$ openocd -c gdb_port 2331 -c telnet_port 2333 -f ~/openocd.cfg -c 'echo "Started by GNU ARM Eclipse"'
Open On-Chip Debugger 0.10.0-dev-00189-g554313b (2016-01-06-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
3333
4444
adapter speed: 8000 kHz
Info : add flash_bank kinetis kl17.flash
adapter speed: 1000 kHz
none separate
cortex_m reset_config sysresetreq
Info : add flash_bank kinetis pflash.0
Started by GNU ARM Eclipse
Info : No device selected, using first device.
Info : J-Link V9 compiled Mar  2 2016 14:43:29
Info : Hardware version: 9.30
Info : VTarget = 1.801 V
Info : clock speed 1000 kHz
Info : SWD IDCODE 0x0bc11477
Info : kl17.cpu: hardware has 2 breakpoints, 2 watchpoints
Info : MDM: Chip is unsecured. Continuing.
`

ilg-ul, 2016-04-05:
yes, that's correct, if you start openocd by hand, you must enclose the entire command you pass to -c in single quotes. 

what is the problem with that?

jacobjennings, 2016-04-05:
Looks like a simple configuration issue with ${openocd_path}, but the error it shows an ambiguous "cannot run program: unknown reason". Copying the command worked because of the $path configuration in my shell.

ilg-ul, 2016-04-05:
I don't get it. If you misconfigured ${openocd_path}, fix it and it should work.

If you are convinced you identified a bug, create a new issue and provide all details to reproduce it.

adroit91, 2017-01-16:
I can confirm that adding the quotes around the "echo <....>" command helps, but how do we fix it within Eclipse? Right now, it's fixed only by launching OpenOCD from outside the IDE.

ilg-ul, 2017-01-16:
Eclipse adds quotes internally.

micheleg83, 2017-11-13:
I managed to reproduce the problem experienced by adroit91. 
The issue is caused by a configuration in "OpenOCD Setup" section of the debug tab in eclipse's debug configuration window. 
If you just specify the executable filename i.e. "openocd" eclipse does not add the proper single quotes, resulting in the error described by adroit91.

To fix this behaviour you can just use the full path to the openocd executable i.e.

 /usr/bin/openocd

ilg-ul, 2017-11-13:
the quotes are necessary only in the echo command. 

openocd should be installed in a separate folder, as explaind in the documentation and the path configured in the MCU Preferences page.

if you do otherwise you're on your own.

I'm afraid you deal with some side effects.

micheleg83, 2017-11-13:
That is correct, the quotes are necessary only in the echo command. To make it work inside eclipse though (without calling openocd from a terminal) it is necessary to specify the full path to the openocd executable (red circle in the image below) in the debug configuration window, at least in my installation, which is now working.

[image]

ilg-ul, 2017-11-13:
> at least in my installation, which is now working.

openocd is guaranteed to work only if you follow (by the book) the instructions from the web.