eclipse-embed-cdt/eclipse-plugins #15: Pack

Gogin79, 2015-10-17:
Hi, I have a problem with update pack. You can not execute refresh package. In the error window is writen : "http://www.keil.com/pack/.......pdsc." Failed to open connection, response code 404. I re-install Eclipse and OFF antivirus. I do not know where I find fault. Please help. Thank you beforhand for your time.

ilg-ul, 2015-10-17:
the keil.com server returns 404 for all .pdsc files

please report the problem to Keil.

Gogin79, 2015-10-17:
9 items added and other items with error. I have 96 clicks items. I did not have more on my nerves :-).

Ok, I write to keil.

ilg-ul, 2015-10-17:
Some of the packages are stored on vendor server, those are available. Most of them are on keil.com, and these fail.

Gogin79, 2015-10-17:
I understand. I wrote to KEIL. I will wait and see what will happen. 

I will write as soon as I get a message from KEIL.

Virtex6, 2015-10-19:
It is working again. Downloaded all the *.pdsc files.

ilg-ul, 2015-10-19:
I confirm.

closed 2015-10-19 by ilg-ul