eclipse-embed-cdt/eclipse-plugins #17: Create a project without newlib and µOS++ III

sparkbuzz, 2015-10-20:
This might be a bit of a silly question, but is it at all possible to build a STM32F4 project without the Newlib and µOS++ III libraries?

I've seen some minimalistic STM32F4 examples that just contain a main.c, and ldscript and some of the HAL libraries, but not sure how to compile them with GNU ARM Eclipse. I'd like an approach where instead of going through __initialize_hardware(), I can have main() be my entry point instead of having µOS++ III initialize the hardware before main is executed.

I'd like to, for example import the STM32F4 Cube examples and compile them with GNU ARM Eclipse, but not entirely sure how.

ilg-ul, 2015-10-20:
> build a STM32F4 project without...

sure, create the project completely empty, without inits and libraries, define your own startup code and jump to main().

after you do it once, you'll learn to appreciate the amount of work put in the existing templates that you want to scrap.

> import the STM32F4 Cube examples

if the Cube examples are makefile projects, you can import them as any other makefile projects.

if you want to import them as managed projects, you need to configure the source folders, the include folders, the symbols and compiler settings.

but you need some experience with Eclipse CDT.

postspectacular, 2015-10-20:
> if you want to import them as managed projects, you need to configure the source folders, the include folders, the symbols and compiler settings.
>
> but you need some experience with Eclipse CDT.

It'd be incredibly helpful if there could be at least some pointers to possibly existing tutorials how to achieve this conversion. I've been wasting hours trying to manually reconstruct some of the STM32F4Cube projects by copying files... but there's got to be a better way! Alas not much info to be found for setting up virtual source folders and configure exclusion settings for CDT. Would it maybe make sense to create a ticket specifically for this use case? Thanks Liviu!

sparkbuzz, 2015-10-20:
Thanks for the repy @ilg-ul.

I'm already appreciating the work done with the templates. Without it, I would have given up learning ARM development a long time ago. ;)

It's just a tad difficult trying to get the STM32F4Cube projects running out of the box. As part of the learning process, I'm also wanting to learn how to setup a STM32F4 project from scratch. Once I understand the basics of ldscripts and how the startup stuff works before main(), I'll be more at ease with using templates.

As for CubeMX projects, when I configure my project, I keep the changes aside, git tag them and diff out the changes made and I manually apply them to my project.

This way I "know my code", and aren't left completely to the mercies of code generators. Which is also why I like GNU ARM Eclipse, no weird wizards trying to take over the whole project.

sparkbuzz, 2015-10-20:
Where can I read more about the newlib startup lifecycle?

ilg-ul, 2015-10-20:
>  the newlib startup lifecycle?

actually there is no such thing as a _newlib startup lifecycle_, in other words there is nothing in newlib that requires special inits (maybe except some file descriptors when using semihosting, but this is a single call).

the startup sequence I used in my projects is quite standard: the data is copied, the bss section is initialised, constructors are executed, arguments are prepared, `main()` is called, if returned the processor soft reset is used (or the semihosting exit routine is called with the exit code).

the only embedded specific thing is the code to initialise the hardware, split into two routines, one executed very early, intended to activate RAM, if needed, and one after the data & bss are initialised.

> Where can I read more 

the best place to read more is the source code itself. in the debug configuration, move the initial breakpoint from `main` to `_start` and you'll be able to step through the entire startup sequence. actually, to save time, I suggest to avoid stepping through the loops, it is much better to place breakpoint over the loops.

postspectacular, 2015-10-20:
@sparkbuzz in addition to @ilg-ul's comments, the official newlib docs are here: https://sourceware.org/newlib/libc.html

ilg-ul, 2015-10-20:
> existing tutorials how to achieve the conversion to managed projects

I don't know, but there probably are some, ask Google.

but things are really simple, by default managed projects just compile all source files found in the project.

if you need more control, it is recommended to remove the project folder from the list of source folders, and add a more specific one, like `src`. the build plug-in will parse all folders in the list of source folders and compile all files that can be compiled.

if the include headers are located in the project folder, they are automatically found by the compiler. if the headers are in separate folders, like `include`, you need to add these folders to the list of included folders.

so for _civilised_ projects, you can easily convert them to managed by defining the source folders and include folders.

ilg-ul, 2015-10-20:
> As for CubeMX projects, when I configure my project, I keep the changes aside, git tag them and diff out the changes made and I manually apply them to my project.

generally you can automate this using the three-way-merge feature, also available in git.

for this you start in the empty repository with two branches with a common ancestor, one with originals, one with your changes, usually master. from time to time you update the originals, then merge them into the master.

an example of a repository using this approach can be the newly created [xpacks/stm32f4-cmsis](https://github.com/xpacks/stm32f4-cmsis), where I plan to maintain a much smaller STM32F4 pack, but still be able to update when ST issues a new release.

ilg-ul, 2015-10-20:
> Would it maybe make sense to create a ticket specifically

yes, this topic is already too complex, I suggest you generally open separate tickets for separate topics.

sparkbuzz, 2015-10-21:
Thanks for all the feedback. GNU ARM Eclipse is awesome, kudos to the contributors for all your efforts put into making ARM development easier!

closed 2015-10-26 by ilg-ul