eclipse-embed-cdt/eclipse-plugins #19: How to use C style directives in assembler ?

weiguixm, 2015-10-22:
I'm reading u-boot sourc code recently. I found u-boot uses C style directives such as "#define" "#ifdef"  and etc in assemble files. But  the assembler only recognize directives like ".ifdef" ".define" by default.  I guess there is some command line options of the assembler to make it recognize C style directives instead of the default ones. It will be appreciate if some one could tell me how to deal with this issue.

ilg-ul, 2015-10-22:
If you are building managed projects with the plug-in, it only accepts assembly files with upper case `.S` extension, which are not compiled with AS, but with GCC, which automatically runs the source code through the preprocessor.

So all you have to do is be sure the file names end with `.S`, the rest is handled automatically.

weiguixm, 2015-10-22:
thank you very much!  I've found the key of the problem. I do the assembling in windows ,but when I extract the tar file of u-boot windows will change the ".S" extension to lower case automatically.

ilg-ul, 2015-10-22:
ok, good luck with your projects!

closed 2015-10-26 by ilg-ul