eclipse-embed-cdt/eclipse-plugins #20: Packs Update is extremely slow

HansHamster, 2015-10-22:
Every time I click the update button, it processes several packs at reasonable speed (several seconds at most) and then grinds to a complete halt for a random pack.
This happens with the current release and the beta of the plugin for both cached files and freshly downloaded ones.
Even canceling the update can take over 10 minutes:

```
[...]
Copy 79kB
 from "http://www.keil.com/pack/Keil.LPC1700_DFP.pdsc"
 to   "/home/eric/Packages/.cache/Keil.LPC1700_DFP.2.1.0.pdsc"
Processing "/home/eric/Packages/.cache/Keil.LPC1700_DFP.2.1.0.pdsc" for content.xml ...
Schema version "1.2".
Completed in 5538ms.
Copy 65kB
 from "http://www.keil.com/pack/Keil.LPC1800_DFP.pdsc"
 to   "/home/eric/Packages/.cache/Keil.LPC1800_DFP.2.5.0.pdsc"
Processing "/home/eric/Packages/.cache/Keil.LPC1800_DFP.2.5.0.pdsc" for content.xml ...
Schema version "1.3".
Completed in 16024ms.
Copy 25kB
 from "http://www.keil.com/pack/Keil.LPC4000_DFP.pdsc"
 to   "/home/eric/Packages/.cache/Keil.LPC4000_DFP.1.1.0.pdsc"
Processing "/home/eric/Packages/.cache/Keil.LPC4000_DFP.1.1.0.pdsc" for content.xml ...
Schema version "1.2".
Completed in 1744ms.
Copy 70kB
 from "http://www.keil.com/pack/Keil.LPC4300_DFP.pdsc"
 to   "/home/eric/Packages/.cache/Keil.LPC4300_DFP.2.5.0.pdsc"
Processing "/home/eric/Packages/.cache/Keil.LPC4300_DFP.2.5.0.pdsc" for content.xml ...
Schema version "1.3".
Completed in 979908ms.
Job cancelled
```

I left it running for several hours without any visible progress. It stops on every try, but the affected pack seems to be completely random.

System:
Eclipse 4.5.1
Linux 4.2.2
openjdk 1.8.0_65

ilg-ul, 2015-10-22:
There were no changes in the packs plug-ins. To be sure, you can try an older version.

However last weekend Keil replaced the servers, so it might be a problem there.

ilg-ul, 2015-10-26:
is it ok now?

closed 2015-12-24 by ilg-ul