eclipse-embed-cdt/eclipse-plugins #21: Unable to find Packs using Package Manager

lalit0611, 2015-10-25:
I followed the instruction as precisely as i could. I have tried refreshing the Packs several times, but no success. This is the warning in the console. Also I checked the whole console, there was no error related to downloads or anything. Please help. Thanks in advance :)
```
Update packs completed in 196s.
2015-10-25 19:35:17
Loading repos summaries...
Parsing cached content file "/home/katph/c_workspace/packages/.cache/.content_www_keil_com_pack.xml"...
File does not exist, ignored.
Parsing cached content file "/home/katph/c_workspace/packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"...
File parsed in 1ms.
Identifying installed packages...
Found no installed packages.
Completed in 2ms.

2015-10-25 19:35:17
Collecting devices...
Found none.
Completed in 1ms.

2015-10-25 19:35:17
Collecting packs...
Found none.
Completed in 1ms.
```
**The contents of the xml file**:
```
<?xml version="1.0" encoding="UTF-8"?>

<root version="1.1">
    <repository name="GNUARM">
        <description>GNU ARM Eclipse packs repository</description>
        <properties>
            <property name="type">xcdl.repo</property>
            <property name="repo.url">http://gnuarmeclipse.sourceforge.net/content.xml</property>
            <property name="generator">GNU ARM Eclipse Plug-ins</property>
            <property name="date">20140716090015</property>
        </properties>
    </repository>
</root>
```

ilg-ul, 2015-10-25:
I just installed a new Eclipse with a fresh workspace, and the Refresh button brought the full list of packs, except the WolfSSL, which returned 403.

The **GNU ARM Eclipse packs repository** is indeed empty, but the Keil repository has more than 100 packs there.

I suggest you rename/remove the `Packages` folder, restart Eclipse and try again.

lalit0611, 2015-10-25:
Deleting and renaming the directory worked. Thanks for the solution :+1:

ilg-ul, 2015-10-26:
Hmmm... Normally all you should do is

* close Eclipse
* remove (or rename) the `Packages/.cache` folder
* restart Eclipse

Since the CMSIS Packs use huge XMLs, the Pack Manager caches some relevant data in the `.cache` folder. If, for any reason, this cache gets corrupted, the simple way to fix it is to remove this folder.

If this does not help, the ultimate solution is to remove the entire `Packages` folder.

closed 2015-10-26 by lalit0611

fli79, 2017-02-15:
I am experiencing similar issues: I cannot select devices in the project properties. After digging I found this GNU ARM Eclipse Packs console log after going to [project properties]>[settings] of an example project.

It seems the repos are not downloaded?

> 2017-02-15 22:35:39
> Extracting devices & boards... 
> Loading repos summaries...
> Parsing cached content file "/home/ffi/Packages/.cache/.content_www_keil_com_pack.xml"...
> **File does not exist, ignored.**
> Parsing cached content file "/home/ffi/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"...
> **File does not exist, ignored.**
> Identifying installed packages...
> Found no installed packages.
> Completed in 101ms.
> No installed packages.
> Completed in 129ms.

- Xubuntu 14.04
- Eclipse: After having Eclipse Neon.2 installed, and after reading http://gnuarmeclipse.github.io/plugins/install/ I have installed eclipse Mars.2 - did not solve the issue.
- Installed the Eclipse plug-ins through the marketplace
- Packs version 2.2.2.2017.01141

Removed the ~/Packages directory and restarted - did not work.

Any ideas what to do?

ilg-ul, 2017-02-15:
are you sure you installed any package?

fli79, 2017-02-16:
No - I did not install any package :( 
I followed this page to install packages: http://gnuarmeclipse.github.io/plugins/packs-manager/#the-packs-perspective)
And then I could indeed select the device in the project properties.

I went on with http://gnuarmeclipse.github.io/tutorials/blinky-arm/ 
and got virtual LED's blinking! 👍

ilg-ul, 2017-02-16:
> No - I did not install any package :(

for the next version I'm planning some AI features to guess what packages you have in your mind and install them automatically. ;-)

hoangtritam, 2020-10-10:
Hello,

I got the same issue here in Eclipse 2020-09. But the problem is due to the URL of Keil package.
It should be provided in HTTPS: **https**://www.keil.com/pack/index.pidx.
The file was now downloaded as the HTTP URL was redirected.

Regards,
Tri Tam

ilg-ul, 2020-10-11:
The code was updated to use https, but it shows like this only when you create a new workspace, if you don't change it manually, older workspaces will remember the http url.