eclipse-embed-cdt/eclipse-plugins #28: I can't choose a device for my project.

xiaoyaojiujian, 2015-10-26:
"Not yet used during build!" how to handle it?
[image]

xiaoyaojiujian, 2015-10-26:
I follow the tutorials to do,http://gnuarmeclipse.github.io/eclipse/project/assign-device/
But always show errors,"Not yet used during build"
how to assign device?

ilg-ul, 2015-10-26:
> how to assign device?

simply select the desired device.

the "Not yet used during build" message informs you that the selection is not yet used during builds, only the debuggers consider it, to select the peripheral registers.

xiaoyaojiujian, 2015-10-27:
Thank you for your answer.
More detailed introduction about these steps?
This tool is different from "Keil".You just need to select the device type, write code, and then directly debug.You don't need to worry about other things, only cares about the code.
 I'm sorry, I'm a newbie.

ilg-ul, 2015-10-27:
all available documentation is at http://gnuarmeclipse.github.io/

good luck with your projects!

ilg-ul, 2015-12-13:
For such questions, in the future please use the GNU ARM Eclipse forum, available at http://www.element14.com/community/groups/gnu-arm-eclipse.

closed 2015-12-24 by ilg-ul