eclipse-embed-cdt/eclipse-plugins #30: How is the /system/ directory excluded from Eclipse's indexer/analyzer?

bitsmack, 2015-10-27:
Hello,

I have been using the gnuarmeclipse plug-ins for some time, developing on the STM32F107 using c. The tools are excellent. Thank you.

Using the default STM32F10x template, it appears that everything in the /system folder (including CMSIS, the standard peripheral libraries, etc) is being excluded from Eclipse's Code Analyzer.

Sometimes Eclipse will have a problem and I'll get a bunch of warnings from one or more of the files in /system. I can get rid of these errors by rebuilding the Eclipse index and then restarting Eclipse.

Now, however, I have warnings that I cannot make go away.  Please see the screenshot. These are spawned by Eclipse's CODAN, not by the gcc toolchain. These just started showing up. I have a different project built with the same template that doesn't show these errors.

[image]

Do you have any ideas for me? I do not want to disable the code analyzer entirely.

Thank you.

Versions:
- Plug-in: Core 2.7.1; 
- Eclipse: Luna SP2 (4.4.2)
- Java: 1.8.0_65
- OS: Win7 Pro, XP1, 64-bit
- Toolchain: 4.9.3

ilg-ul, 2015-10-27:
> excluded from Eclipse's Code Analyzer.

the plug-ins do not explicitly exclude anything from the indexer/analyser. actually they ignore the analyser entirely, and this might be the problem, perhaps there are some settings required for the analyser to work properly, settings that are not performed by the template project wizard.

unfortunately I have no experience with the code analyser, so if you discover how to fix the projects, I can update the wizards.

I suggest you start with very simple projects, created with the hello world template, and, if they work properly, increase complexity, checking differences in the .project/.cproject files, in the gui settings, etc.

bitsmack, 2015-10-28:
Thank you for your quick comment.

This is interesting because _only occasionally_ will the analyser show warnings from a subset of files in the spl, or in the cmsis directories.  As I mentioned, these warnings stop being displayed after a re-index and then an Eclipse restart...  It doesn't seem to be the same files or the same types of warnings. And when these warning are _not_ being displayed, I still get warnings on files in my /src and /include directories.

Makes me wonder why the analyser usually _doesn't_ display the warnings!

It'll be a few weeks before I have time to work on it further.  I expect you'll want to close this issue in the mean time.  If I discover anything, would you prefer an email or should I create a new Issue? (Or, can I comment on a closed issue?)

Thanks again.

ilg-ul, 2015-10-28:
please check if you can comment on a closed issue. if so, you can close this issue and later comment on it and you or me can reopen it. if not, we can leave the issue open.

bitsmack, 2015-10-30:
OK, I was able to post on a closed issue.  I'll close this one, and come back later after I've done some research. Thanks for everything.

closed 2015-10-30 by bitsmack