eclipse-embed-cdt/eclipse-plugins #31: Add Segger Real Time Terminal (RTT) support to J-Link plug-in

ilg-ul, 2015-10-28:
As a faster alternative to semihosting, [RTT](https://www.segger.com/jlink-real-time-terminal.html) provides a bidirectional channel between the target and the host, that can be used to send trace messages.

metc, 2016-05-26:
Yes, it would be great if [the RTT terminal](https://www.segger.com/jlink-rtt.html) could be included in Eclipse. I currently use the CF Terminal (Console) View plugin, as described here: https://mcuoneclipse.com/2015/04/20/serial-terminal-view-in-eclipse-luna/

ilg-ul, 2016-05-26:
I am still considering it, but still had no time to investigate how complicated the implementation might be.

gonzzor, 2018-11-07:
Any more ideas on this? The RTT Console is basically just a Telnet connection to JLinkGDBServer. Another option would be to launch JLinkRTTClient.

ilg-ul, 2018-11-07:
as long as there are workarounds, the priority will probably remain low.