eclipse-embed-cdt/eclipse-plugins #33: Support for Linaro AArch64 bare-metal toolchain version 4.9.

iulia-vasii, 2015-11-03:
Please note that new Linaro tool provided for Baremetal 64bit has been renamed from aarch64-none-elf-g++ into aarch64-elf-g++, thus an update is required.

Question: is there a way to configure the prefix for each toolchain similar with what we have for toolchain's path/name, explained here: http://gnuarmeclipse.github.io/developer/preferences/ilg.gnuarmeclipse.managedbuild.cross?

ilg-ul, 2015-11-03:
> aarch64-elf-g++

updated, please check the version from updates-test.

> configure the prefix for each toolchain

as far as I remember, these definitions are currently static in the plug-in, do not come from preferences.

iulia-vasii, 2015-11-04:
I tested but the fix is not complete, the prefix in the Toolchains tab should also be updated, not only the name.

Do you have any idea how the backward compatibility could be maintained? The new plug-in and the Linaro toolchains < 4.9 will no longer work together after this update.

A very useful feature would be to give the chance to developer users to add new toolchains from code (using an extension point for instance). Can I log this separately?

ilg-ul, 2015-11-04:
> the prefix in the Toolchains tab should also be updated

the updated toolchain definition will be used for new projects. existing projects will continue to use their definitions stored in the `.cproject` file, until you update them (for example by selecting another toolchain and returning to the desired one).

> The new plug-in and the Linaro toolchains < 4.9 will no longer work together after this update.

that's correct.

if you want to define and implement an extension point to add custom toolchains, I can consider it for integration into the plug-ins.

iulia-vasii, 2015-11-05:
That's right.

Ok, I can try to implement that.

ilg-ul, 2015-11-06:
added to 2.10.2-201511061603

closed 2015-11-06 by ilg-ul

iulia-vasii, 2015-12-10:
Could you please do the same for "Linaro AArch64 big-endian bare-metal ELF" toolchain?

ilg-ul, 2015-12-10:
can you be more specific what exactly you need?

iulia-vasii, 2015-12-10:
Update the prefix for "Linaro AArch64 **big-endian** bare-metal ELF" from "aarch64_be-none-elf-" to "aarch64_be-elf-". If you remember, this change was done recently for the little endian toolchain (Linaro AArch64 bare-metal ELF).

ilg-ul, 2015-12-11:
done. please check updates-test.

next time please use pull requests.

iulia-vasii, 2015-12-11:
It's working, thanks you.
I'll try to use pull requests next time (I'm not familiar with github process).

Any idea when the next release will be announced?

ilg-ul, 2015-12-11:
> not familiar with github process

it's easy, you clone the repo to your account, create a branch, make the changes, submit the pull request, I merge it to the main repo.

https://help.github.com/articles/what-is-a-good-git-workflow/

> next release

I hope soon, the only remaining issue is the Restart button, which does not work on Mars.

iulia-vasii, 2015-12-11:
Ok, thanks for the info.