eclipse-embed-cdt/eclipse-plugins #34: Help Wanted

marcandrelm, 2015-11-07:
I installed the plug-in using the http address in eclipse but I cannot see any of the gnu arm toolchain when I create a new projet. Any help or guidance will be appreciated

Here are some system spec:
-Window 10 x64
-GNU ARM C/C++ Cross Compiler 1.15.2.201511061603 and 
all of the other packages of 7th november 2015.
-Eclipse : Mars.1 Release (4.5.1)
-java version "1.8.0_66" and "1.7.0_79"
-MinGW : gcc (GCC) 4.8.1
-GNU Tools ARM Embedded 4.9 2015q3
-GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00141-g09aeb96-dirty

ilg-ul, 2015-11-07:
> I cannot see any of the gnu arm toolchain when I create a new project.

where do you expect to see the toolchain? you just need to enter/check the toolchain path.

marcandrelm, 2015-11-07:
What i mean is the menu seen on this page : http://gnuarmeclipse.github.io/tutorials/hello-arm/ 
the image above "In the  Target processor settings window be sure to:"
I do not see any of the exceutable templates like the STM32F0xx C/C++ project or the Hello World ARM example.

ilg-ul, 2015-11-07:
> I do not see any of the executable templates like ...

then what do you see? are you sure all Eclipse plug-ins were correctly installed (check Eclipse -> About -> Installation Details -> Features for the ilg.gnuarmeclipse.templates.stm feature)

marcandrelm, 2015-11-07:
I only see the starting option of "Empty Project" and "Hello World ANSI C Project". in the Exceutable folder in the C project wizard. All of the plug-ins are there from ADuCM360 Project Template to the STM32Fx Project.

I had an other issue I think. The toolchain that I had was not linked to Eclipse properly. 
I removed it and redoing the process on 
http://gnuarmeclipse.github.io/windows-build-tools/install/
But I connot find the Global Tools Paths onf Eclipse Mars 1.
(the image above "The above definition will make the build tools accessible to all projects in all workspaces.")

ilg-ul, 2015-11-07:
there is no relationship between the windows build tools, the presence of the toolchain and the content of the create c/c++ project wizard or the global tools paths preferences window.

it looks like you have a broken eclipse. better reinstall it from scratch, following the instructions by the book.

marcandrelm, 2015-11-07:
I had that issue with other plug-ins before.
Maybe the plug-ins have to be rebuild for Mars?
I'll check into that after I try the nuclear option on Eclipse.

ilg-ul, 2015-11-07:
> Maybe the plug-ins have to be rebuild for Mars?

nope

marcandrelm, 2015-11-07:
The nuclear option seems to work.
I'll keep you updated if I can go up to debug with OCD.

marcandrelm, 2015-11-08:
Having issues with Jlink and Qemu but the eclipse is good now.
Thanks for the help Ilg.

closed 2015-11-08 by marcandrelm

ilg-ul, 2015-12-13:
For such questions, in the future please use the GNU ARM Eclipse forum, available at http://www.element14.com/community/groups/gnu-arm-eclipse.