eclipse-embed-cdt/eclipse-plugins #35: Trace Control for GDB

hackrid, 2015-11-10:
it would be great to be able to control the trace peripheral from within Eclipse.

There might be some button to start program counter sampling.
https://github.com/PetteriAimonen/STM32_Trace_Example/blob/master/configure-trace.openocd
I have no idea how to add this to eclipse :(
a second tool could then parse the PC samples to a statistical report about the time the processor spends where.

ilg-ul, 2015-11-10:
yes, I investigated, enabling trace is not very difficult, processing and displaying the result is the big problem.

another problem is that not all cores have the memory buffers to store the samples, and the alternative, to use the multi-bit trace interface requires expensive probes.

my priority would be:
* get the events (interrupt entry/exit), 
* get data trace (monitor ram locations that change)
* get the instruction trace.

it is on my TODO list, but unless someone else will support the development, it has low priority.

hackrid, 2015-11-11:
Hi Liviu,

thanks for your quick answer.
I agree on the difficulties to display the trace information. Have you seen any charting solutions that might fit the needs?
as far as I understand, for most (hardware) tracing, no buffers are needed since the information is streamed out via TPIU and the core gets stalled once to FIFO is full

what would you suggest on how to configure the trace functionality?

two possible solutions I think of are:

* python gdb scripts writing to the trace registers in the processor?
   - this would not need code changes on the target but maybe lots of different scripts on the host to configure the trace registers
* a processor depended trace library which provides a set of unified function names across processors to configure the trace unit
   - to start PC sampling you may could write "call gnuarm_trace_start_pc_samping" in your gdb console and have the plattform specific action  executed on the target.

maybe the multi-bit trace interface could also be accessed with a logic analyzer like it is shown here:
http://essentialscrap.com/tips/arm_trace/theory.html

one then would not need a expensive debugger but a logic analyzer software with live decode functionality. don't know is sigrok supports that.

ilg-ul, 2015-11-11:
> Have you seen any charting solutions that might fit the needs?

yes, I liked the way the former CodeRed displayed events and data trace. I guess LPCExpress inherited these features too, but I did not test them.

> what would you suggest on how to configure the trace functionality?

I would check the information available in SVD + PDSC and directly set the registers. 

the setting should happen in the debugging plug-in, not in the application.

hackrid, 2015-11-11:
this describes a way to decode trace information with sigrok (open source logic analyzer software) continously:
http://www.drdobbs.com/embedded-systems/rocking-with-sigrok/240147198

ilg-ul, 2015-11-11:
J-Link already does part of the job. you can get the trace stream via a special tcp port, and decode it yourself, or let the server decode it and print the result on the console.

as it is now implemented, the, the ITM trace is processed by J-Link and printed in the console; it works quite fine.

please check the SEGGER manual for more details.

hackrid, 2015-11-11:
i was mentioning the logic analyzer approach to maybe solve the expensive hardware problem with multi-bit tracing.

I have to check the SWO output on the Segger Terminal once I have access to a STM board.

ilg-ul, 2015-11-11:
> the logic analyzer approach

this might work, but I don't consider it a professional approach.

> SWO output ... STM board

I think SWO might be available on other MCUs too.

hackrid, 2015-11-12:
is the multi-bit data from a J-Trace probe avaiable somehow via GDB ? I could not find a statement in the Documents.

ilg-ul, 2015-11-12:
> multi-bit data from a J-Trace

I did not try it, but I guess you can access by telnetting to the swo tcp port.

hackrid, 2015-11-12:
wow yes. the lpc tracing features look promesing.

how can these plugins be extracted and used in a bare eclipse with segger gdb ?

[image]

ilg-ul, 2015-11-12:
> how can these plugins be extracted and used in a bare eclipse with segger gdb ?

I doubt they were designed for this.