eclipse-embed-cdt/eclipse-plugins #36: Restart button not working in Eclipse Mars

enz-ml, 2015-11-17:
After switching to Eclipse Mars i noticed that the restart button in the Debug perspective stopped working. I can see in the log-window that "something" is happening (see log below). But the processor doesn't start over. It seems, that the processor just starts running where it was stopped.

The failing system is:
Eclipse: Version: Mars.1 Release (4.5.1) Build id: 20150924-1200
CDT: 8.8.0.201509131935
Gnu Arm Eclipse: 3.2.1.201511061603
Win 7 Pro 64 bit, but same behaviour in Linux Mint 17.1 64 bit under Virtual Box
ToolChain: GNU Tools ARM Embedded\4.9 2015q1 and also GNU Tools ARM Embedded\4.9 2015q2

The working system is:
Eclipse: Version: Luna Service Release 2 (4.4.2) Build id: 20150219-0600
CDT: 8.6.0.201502131403
Gnu Arm Eclipse: 3.2.1.201504061754
Win 7 Pro 64 bit
ToolChain: GNU Tools ARM Embedded\4.9 2015q1 and also GNU Tools ARM Embedded\4.9 2015q2

I tried it with Segger-Jlink Software V4.98e and V5.02g. Both behave the same under Luna and Mars, So this doesn't seem to be the culprit.

This is the log of the failing setup (Mars) after the processor was already halted:
Read 2 bytes @ address 0x08006A2A (Data = 0xF7FF)
Read 2 bytes @ address 0x08006A2A (Data = 0xF7FF)
Read 2 bytes @ address 0x08006A2A (Data = 0xF7FF)
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 20000BA0
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000, R13= 20000B98, MSP= 20003FA8, PSP= 20000B98
R14(LR) = 080030B5, R15(PC) = 08006CEA
XPSR 01000000, APSR 00000000, EPSR 01000000, IPSR 00000000
CFBP 02000000, CONTROL 2000000, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x08006CEA (Data = 0x46BDAF00)
Setting breakpoint @ address 0x08006A2A, Size = 2, BPHandle = 0x0006
Starting target CPU...

And this is the log of the working setup (Luna) after the processor was already halted:
Resetting target
Halting target CPU...
...Target halted (PC = 0x080001DC)
Read 2 bytes @ address 0x0800189A (Data = 0xF7FF)
Read 2 bytes @ address 0x0800189A (Data = 0xF7FF)
Read 2 bytes @ address 0x08001898 (Data = 0xB508)
R0 = FFFFFFFF, R1 = FFFFFFFF, R2 = FFFFFFFF, R3 = FFFFFFFF
R4 = FFFFFFFF, R5 = FFFFFFFF, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF, R13= 20002000, MSP= 20002000, PSP= FFFFFFFC
R14(LR) = FFFFFFFF, R15(PC) = 080001DC
XPSR C1000000, APSR C0000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x080001DC (Data = 0xF7FFB508)
Setting breakpoint @ address 0x08001898, Size = 2, BPHandle = 0x0002
Starting target CPU...
...Breakpoint reached @ address 0x08001898
Reading all registers
Read 4 bytes @ address 0x08001898 (Data = 0xF7FFB508)
Removing breakpoint @ address 0x08001898, Size = 2

I don't have any clue how to search further. Any hint?

Regards,
Martin

ilg-ul, 2015-11-17:
I'll investigate, but from your logs I see the two tests are completely different. Could you redo the test and build exactly the same test, with the same toolchain, and the same options?

enz-ml, 2015-11-18:
To make the systems identical, I updated Eclipse Luna:

Luna Service Release 2 (4.4.2) Build id: 20150219-0600
CDT: 8.6.0.201502131403
Gnu Arm Eclipse: 3.2.1.201511061603
ToolChain: GNU Tools ARM Embedded\4.9 2015q2
Win 7 Pro 64 bit

And after that, I noticed the same behaviour as on Eclipse Mars.
So maybe this is related to the version 3.2.1.201511061603 of Gnu Arm Eclipse?
The sequence was:
Create blinky example with template for STM32F051R8 (STM32F0 discovery board was used).
Compile, Create debug configuration,
Set breakpoint at line 81 (timer_sleep(BLINK_OFF_TICKS);)
Debug, Run (breakpoint will be hit), Click restart button
Here is the complete log:

SEGGER J-Link GDB Server V5.02g Command Line Version

JLinkARM.dll V5.02g (DLL compiled Oct 26 2015 18:58:50)

-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F051R8
Target interface:              SWD
Target interface speed:        0kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link Pro V4 compiled Oct  8 2015 09:29:10
Hardware: V4.00
S/N: 174301211
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 2.98 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000000 (Data = 0x20002000)
Target interface speed set to 30 kHz
Resetting target
Halting target CPU...
...Target halted (PC = 0x080001DC)
R0 = FFFFFFFF, R1 = FFFFFFFF, R2 = FFFFFFFF, R3 = FFFFFFFF
R4 = FFFFFFFF, R5 = FFFFFFFF, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF, R13= 20002000, MSP= 20002000, PSP= FFFFFFFC
R14(LR) = FFFFFFFF, R15(PC) = 080001DC
XPSR C1000000, APSR C0000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x080001DC (Data = 0xF7FFB508)
Select auto target interface speed (2000 kHz)
Flash breakpoints enabled
Read 4 bytes @ address 0x080001DC (Data = 0xF7FFB508)
Downloading 536 bytes @ address 0x08000000 - Verified OK
Downloading 40 bytes @ address 0x08000218 - Verified OK
Downloading 1676 bytes @ address 0x08000240 - Verified OK
Downloading 36 bytes @ address 0x080008CC - Verified OK
Writing register (PC = 0x080000c4)
Read 4 bytes @ address 0x080000C4 (Data = 0xB083B530)
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000700 (Data = 0xB510)
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000730 (Data = 0x20FA)
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000700 (Data = 0xB510)
R0 = FFFFFFFF, R1 = FFFFFFFF, R2 = FFFFFFFF, R3 = FFFFFFFF
R4 = FFFFFFFF, R5 = FFFFFFFF, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF, R13= 20002000, MSP= 20002000, PSP= FFFFFFFC
R14(LR) = FFFFFFFF, R15(PC) = 080000C4
XPSR C1000000, APSR C0000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x080000C4 (Data = 0xB083B530)
Setting breakpoint @ address 0x08000700, Size = 2, BPHandle = 0x0001
Setting breakpoint @ address 0x08000730, Size = 2, BPHandle = 0x0002
Starting target CPU...
...Breakpoint reached @ address 0x08000700
Reading all registers
Read 4 bytes @ address 0x08000700 (Data = 0xF7FFB510)
Removing breakpoint @ address 0x08000700, Size = 2
Removing breakpoint @ address 0x08000730, Size = 2
Setting breakpoint @ address 0x08000730, Size = 2, BPHandle = 0x0003
Starting target CPU...
...Breakpoint reached @ address 0x08000730
Reading all registers
Read 4 bytes @ address 0x08000730 (Data = 0xF7FF20FA)
Removing breakpoint @ address 0x08000730, Size = 2
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000702 (Data = 0xF7FF)
Read 2 bytes @ address 0x08000700 (Data = 0xB510)
R0 = 48000800, R1 = 00000200, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF, R13= 20001FE0, MSP= 20001FE0, PSP= FFFFFFFC
R14(LR) = 08000731, R15(PC) = 08000730
XPSR 21000000, APSR 20000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x08000730 (Data = 0xF7FF20FA)
Setting breakpoint @ address 0x08000700, Size = 2, BPHandle = 0x0004
Performing single step...
...Target halted (PC = 0x08000732)
Reading all registers
Read 4 bytes @ address 0x08000732 (Data = 0xFFCBF7FF)
Setting breakpoint @ address 0x08000730, Size = 2, BPHandle = 0x0005
Starting target CPU...
...Breakpoint reached @ address 0x08000730
Reading all registers
Read 4 bytes @ address 0x08000730 (Data = 0xF7FF20FA)
Removing breakpoint @ address 0x08000700, Size = 2
Removing breakpoint @ address 0x08000730, Size = 2

ilg-ul, 2015-11-18:
I confirm the problem, on Mars 1 the Restart button does not work, it just resumes execution.

enz-ml, 2015-11-27:
Thanks for the confirmation.
Do you already have a schedule for the next release?

ilg-ul, 2015-11-27:
Next week, but currently I don't have a solution for the Restart button. If I'll find one these days I'll include it, otherwise I'm afraid I have to postpone it to the next release.

enz-ml, 2015-11-28:
That would be a pity.
I really miss that functionality at the moment.

As i said in my last post, It does not seem to be related to MARS 1.
It also doesn't work in Luna, since I updated the Gnu Am Eclipse Plugin to the v2.10.2-201511061603 release.

Martin

ilg-ul, 2015-12-15:
I'm a bit confused, I finally had some time to debug this issue but... I can not reproduce the problem; I tried on two different Eclipse Mars.1, both with QEMU and J-Link, and the Restart button if functional:

For example this is the JLinkGDBServer console:

```
Resetting target
Halting target CPU...
...Target halted (PC = 0x080002B8)
Read 2 bytes @ address 0x08001CE0 (Data = 0xF7FF)
Read 2 bytes @ address 0x08001CE0 (Data = 0xF7FF)
Read 2 bytes @ address 0x08001CD8 (Data = 0xE92D)
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000, R13= 20020000, MSP= 20020000, PSP= 00000000
R14(LR) = FFFFFFFF, R15(PC) = 080002B8
XPSR 01000000, APSR 00000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x080002B8 (Data = 0xF7FFB508)
Setting breakpoint @ address 0x08001CD8, Size = 2, BPHandle = 0x002A
Setting breakpoint @ address 0x08001DA0, Size = 2, BPHandle = 0x002B
Starting target CPU...
...Breakpoint reached @ address 0x08001CD8
Reading all registers
Read 4 bytes @ address 0x08001CD8 (Data = 0x41F0E92D)
Removing breakpoint @ address 0x08001CD8, Size = 2
Removing breakpoint @ address 0x08001DA0, Size = 2
```

Similarly, for QEMU I get:

```
Execute 'mon system_reset'.

Cortex-M4 r0p0 core reset.
PRIGROUP unimplemented
```

I use 

Version: Mars.1 Release (4.5.1)
Build id: 20150924-1200

Any suggestions on how to proceed?

ilg-ul, 2015-12-15:
Could you retry, preferably using a freshly installed Eclipse, the latest plug-ins (available from updates-test), and a new workspace?

enz-ml, 2015-12-16:
Thanks for investigating. I did the recommended test:
New Eclipse download, updates-test and new workspace.
And that combination works:
Version: Mars.1 Release (4.5.1)
Build id: 20150924-1200
Gnu Arm Eclipse:
Version: 3.2.1.201512141335

Did you fix something in updates-test?
What about your setup when you confirmed the bug?

I don't know anything about QEMU, so i can not comment on this.

Glad that it works now, but i am a little bit confused at the moment.

Anything more i should/could test?

ilg-ul, 2015-12-16:
> Did you fix something in updates-test?

nothing related to the debugging plug-ins.

> I don't know anything about QEMU

it takes just a few minutes to follow the tutorial: http://gnuarmeclipse.github.io/tutorials/blinky-arm/

> What about your setup when you confirmed the bug?

it was either very late in the night and I did not know what doing, or there is a possible configuration when Restart does not work.

the debugging plug-ins suffer from interferences between projects, so the tests must be done with a single open project in the workspace.

enz-ml, 2015-12-16:
> the Eclipse debugging plug-ins suffer from interferences between projects, so the tests must be done > with a single open project in the workspace.

So maybe something strange with my workspace?
To be honest, I am surprised that it seems that I am the only one that reports this bug. Because I consider the restart button an often used feature.

I will do some more tests with different versions.

ilg-ul, 2015-12-16:
> I will do some more tests with different versions.

yes, thank you.

if nothing significant happens, I'll prepare the new release tomorrow evening.

enz-ml, 2015-12-17:
It is definitely related to my workspace.
It tried two different new installs of Eclipce Mars.
With the first one, I installed your plugin via the Marketplace (nice option, by the way) and with the other MARS I installed the plugin via the udate site (non test). Both worked fine, with a new, clean.workspace. But as soon as I used my workspace, the restart button stopped working.

Can there something be broken in my workspace?
Anything else I could test?

ilg-ul, 2015-12-17:
> definitely related to my workspace

ok, then I proceed with the release

> as soon as I used my workspace, the restart button stopped working

once something goes wrong and the workspace becomes corrupted, usually it is easier to dispose than fix. this does not mean it can't be fixed (in fact most of the files are properties and xmls), just that there are no clear tools and/or procedures to follow.

from my experience, avoid opening a workspace with different versions Eclipses, especially different platform Eclipses.

try to import your project in a new workspace and work there. if you need other plug-ins, install them one by one and check after each install if the Restart button is still functional.

ilg-ul, 2015-12-17:
>  installed your plugin via the Marketplace ... and with the other MARS ... via the update site

there should be no difference, the Marketplace just provides the update site url, the result should be identical.

enz-ml, 2015-12-17:
I only used the eclipse internal update functionality for the workspaces. But only from older versions to newer versions. I never opened a MARS-workspace with Luna, etc.

> there should be no difference, the Marketplace just provides the update site url, the result should be identical.

I can confirm this now :-)

> try to import your project in a new workspace and work there. if you need other plug-ins, install them one by one and check after each install if the Restart button is still functional.

I only use two addtional plugins, Bracketeer and LaunchBar. Meanwhile I installed both in the new workspace and everything is still working.

I will now import my projects in the new workspace, one by one.

closed 2015-12-24 by ilg-ul