eclipse-embed-cdt/eclipse-plugins #37: Peripheral view is empty, why?

teholabs, 2015-11-17:
Short: Nothing appears in Peripheral tab on debug with ATSAMD10/OpenOCD/CMSIS-DAP; why?

Longer:
I have installed the tools on Windows (10). 

Installed:
GDB Hardware 8.8.0
Eclipse 4.5.1
Packs 2.2.2.201511061603
(not in use but QEMU 2.2.2.201511061603)
OpenOCD 0.9.0-201505190955
Build Tools 2.6-201507152002
GNU ARM Tools 4.9 2015q3

Procedure:
Create new C project -> Hello World ARM Cortex M
M0+
1 MHz
16 kB Flash
4 kB RAM
Freestanding
Trace: None
Check some warnings
-Og on debug
newlib nano
Vendor CMSIS name samd10

I only have 1 complier so next to Finish

Using Packs view I have installed the packs for SAND10_DFP (1.1.0)

Project->Properties-> C/C+ Build-> settings-> Devices tab

I select Atmel-> SAM D Series -> SAM D10-> ATSAMD10D14A
I click apply then okay

The RAM amount is correct so there must a file for this microcontroller somewhere. 

I cut and paste a simple blink the LED with pointers to the PORTA device program and a NOP delay function.

Build-> Yes it complies.

*Setup debugger:*
GDB openOCD Debugging:
elf is correct on Main tab
Debug tab, config option:
-f board/atmel_samd10_xplained_mini.cfg (this file I created as a modification from the D20 board file)

GDB client setup:
set mem inaccessible-by-default off

Run with debug configuration:
Result light blinks fine, registers report fine, memory dumps fine if you put it in by hand where to watch.

Peripherals tab is empty. Right clicking provides no options.
View reset does not help.

Suggestions?

ilg-ul, 2015-11-17:
Unfortunately I could not reproduce your problem.

However I installed the SAND10_DFP package and assigned ATSAMD10D14A to one existing project and the Atmel registers are there. So the package is ok.

I guess you did something wrong. Could you follow the [blinky](http://gnuarmeclipse.github.io/tutorials/blinky-arm/) tutorial and confirm that you see the STM32F4 registers when debugging under QEMU.

Then you assign the Atmel device to the same project, and restart the debug session. You should see the Atmel registers now.

teholabs, 2015-11-18:
I confirm the tutorial works.
Changing to D10 breaks the Peripherals for the tutorial on my machine.
Changing to SAMD11 however does not, even on my project, so it must be a corrupt file somehow for just the D10 Pack I guess...

D11 has the same memory map anyway so I can just use that for now. Thanks for your help.

closed 2015-11-18 by teholabs

ilg-ul, 2015-11-18:
> it must be a corrupt file somehow for just the D10 Pack I guess...

remove that package and install it again