eclipse-embed-cdt/eclipse-plugins #38: GDB problem error when debugging u-boot with JLink

weiguixm, 2015-11-22:
I'm debugging u-boot-2015 with GDB and JLink. 
All the processes on assembler code are OK. 
But when jumping to C code GDB can't display the source code correctly. It will displays some codes in other places. But I'm sure the binary code is still executing correctly. 
I've tried two different versions of u-boot(2015.10 and 2015.07), but it appeared the same problem.

ilg-ul, 2015-11-22:
I guess you built u-boot with a makefile. are you sure that makefile used the correct debug options when compiled/linked u-boot?

the debug plug-in is generally guaranteed to work with programs built by the managed build plug-in, which set the debug options more accurately, for makefile projects it is sometimes tricky.

to check this, run a GDB debug session from the terminal, outside Eclipse. if it behaves similarly, you probably do not have the debug info in the executable. if it works, it means there is a difference in the way you invoked GDB manually, and you need to configure Eclipse to start GDB similarly.

weiguixm, 2015-11-22:
Firstly of all, thank you very much for your reply.
This error is the exactly one I got from command-line GDB.
I've tried to change the -O2 and -Os to -O0, but it doesn't work. And there is already -g option in makefile.
I don't know whether there are other changes in makefile can be tried as I don't understand it's structure very much.

ilg-ul, 2015-11-22:
> This error is the exactly one I got from command-line GDB.

and in this case why do you expect the GDB running under Eclipse to behave differently?

regardless how much I would like it, the GNU ARM Eclipse tools cannot make miracles.

please fix your project, and when it behaves properly with the command-line GDB, return to using Eclipse.

ilg-ul, 2015-11-26:
any progress?

weiguixm, 2015-11-26:
Thanks for your concern. As you said this problem have no relationship with GNU ARM Eclipse tools .

I'm analysing the makefiles and script files of u-boot and trying to change the compiling and linking options.

But I got no progress so far.

It seems that all the C codes will make GDB performed abnormal. When the progress run back to assambler codes GDB will  performe normal again.

Maybe I should also diggin by analysing the object files generated by C files by dissasambling them.

Or is there any possibility that c source files contain some special characters that GDB can't understand? 

I'll update this post asap when I got any progress.

ilg-ul, 2015-12-13:
For such questions, in the future please use the GNU ARM Eclipse forum, available at http://www.element14.com/community/groups/gnu-arm-eclipse.

closed 2015-12-24 by ilg-ul