eclipse-embed-cdt/eclipse-plugins #39: Packs (2.2.2.201511061603) not returning any packs on refresh

ianlokejj, 2015-11-22:
plug-in version: GNU ARM C/C++ Packs (Experimental) 2.2.2.201511061603
eclipse version: 4.4.2
java version: 1.7.0_85
os: Ubuntu 14.04
toolchain version: gcc-arm-none-eabi-4_9-2015q3

I was following to tutorial on http://gnuarmeclipse.github.io/plugins/packs-manager/ . When I hit the refresh button to retrieve the available packs, no available packs were returned. 

Attached is the eclipse console log.
[log.txt](https://github.com/gnuarmeclipse/plug-ins/files/41001/log.txt)

I had an internet connection present throughout the tutorial, tried restarting eclipse and re-installing the plug-in. However I was unable to retrieve any packs.

May I know what I have done wrong?

Thank you for your attention.

ilg-ul, 2015-11-22:
apparently the Keil server returned no packages. what does the `.cache/.content_www_keil_com_pack.xml` file contain?

ianlokejj, 2015-11-22:
<?xml version="1.0" encoding="UTF-8"?>

<root version="1.1">
  <repository name="Keil">
    <description>Keil CMSIS packs repository</description>
    <properties>
      <property name="type">cmsis.repo</property>
      <property name="repo.url">http://www.keil.com/pack/index.idx</property>
      <property name="generator">GNU ARM Eclipse Plug-ins</property>
      <property name="date">20151122083709</property>
    </properties>
  </repository>
</root>

ianlokejj, 2015-11-22:
```
<?xml version="1.0" encoding="UTF-8"?>

<root version="1.1">
  <repository name="Keil">
    <description>Keil CMSIS packs repository</description>
    <properties>
      <property name="type">cmsis.repo</property>
      <property name="repo.url">http://www.keil.com/pack/index.idx</property>
      <property name="generator">GNU ARM Eclipse Plug-ins</property>
      <property name="date">20151122083709</property>
    </properties>
  </repository>
</root>
```

ilg-ul, 2015-11-22:
yes, it is empty.

try a `curl -L http://www.keil.com/pack/index.idx`. if this is empty too, you have a network problem, since from my location I get the correct list of packages.

ianlokejj, 2015-11-22:
Running the curl command does give me the list of packages.

[packages.txt](https://github.com/gnuarmeclipse/plug-ins/files/41010/packages.txt)

ilg-ul, 2015-11-22:
this is unusual. did you check the Eclipse log for exceptions?

try again the Refresh button. if it still reports 0 packages, remove the cached file an retry.

ianlokejj, 2015-11-22:
I deleted the cache file and have hit refresh. For once I am receiving a different log which states that there are 136 packs.
```
2015-11-22 17:30:19
Update packs job started.
Parsing "http://www.keil.com/pack/index.idx"...
Contributed 136 pack(s).
Processing "/home/ianlokejj/Packages/.cache/ABOV.CM3_DFP.1.0.0.pdsc" for content.xml ...
Schema version "1.2".
Completed in 829ms.
"XML document structures must start and end within the same entity.", ignored.
Copy 517B
 from "http://gnuarmeclipse.sourceforge.net/packages/content.xml"
 to   "/home/ianlokejj/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"

Update packs completed in 5s.
2015-11-22 17:30:23
Loading repos summaries...
Parsing cached content file "/home/ianlokejj/Packages/.cache/.content_www_keil_com_pack.xml"...
File parsed in 1ms.
Parsing cached content file "/home/ianlokejj/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"...
File parsed in 1ms.
Identifying installed packages...
Found no installed packages.
Completed in 4ms.

2015-11-22 17:30:23
Collecting devices...
Found none.
Completed in 1ms.

2015-11-22 17:30:23
Collecting packs...
Found none.
Completed in 1ms.
```
From the log, it seems that when processing the first element, the following warning appears:
```
"XML document structures must start and end within the same entity.", ignored.
```
Thereafter, I end up having no packages available.

ilg-ul, 2015-11-22:
I don't know what is wrong, but it looks like an out-of-sync something.

Could you:

* close Eclipse
* move the entire Packages folder somewhere else
* start Eclipse 
* retry Refresh

ianlokejj, 2015-11-22:
I tried the above steps but was still unable to retrieve the packages.

I followed the steps from http://sourceforge.net/p/gnuarmeclipse/bugs/163/#2b0c and created a local copy of the idx file. Next only included packages which I required and proceeded to retrieve the packages by hitting the refresh button. I was successful in obtaining the packages.

Thank you for your time in assisting me with the troubleshooting.

closed 2015-11-22 by ianlokejj

flynneva, 2020-01-15:
I am having exactly these same symptoms described on this issue....other than his given solution was the root of this problem ever solved? 

Eclipse Error shown after refresh:
>org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 50; White spaces are required between publicId and systemId.

It looks like the downloaded file might have a typo in the first line but I am not sure?
Eclipse console log:
>2020-01-15 13:11:45
Update packs job started.
Parsing "http://www.keil.com/pack/index.idx"...
Malformed index, no packs contributed.
File "/home/continental/opt/cmsis-packs/.cache/.content_www_keil_com_pack.xml" written.
>
>Update packs completed in 0s.
>
>2020-01-15 13:11:45
Loading repos summaries...
Parsing cached content file "/home/continental/opt/cmsis-packs/.cache/.content_www_keil_com_pack.xml"...
File parsed in 1ms.
Identifying installed packages...
Found no installed packages.
Completed in 2ms.
>
>2020-01-15 13:11:45
Collecting devices...
Found none.
Completed in 1ms.
>
>2020-01-15 13:11:45
Collecting packs...
Found none.
Completed in 1ms.

ilg-ul, 2020-01-16:
> I am having exactly these same symptoms described on this issue

Nope, in your case parsing the index fails.