eclipse-embed-cdt/eclipse-plugins #40: Linker flags not shown for C++ projects

joelm-somnium, 2015-11-24:
For C++ projects, the 'Linker flags' box is not shown at:
Project properties > C/C++ Build > Settings > Cross ARM C++ Linker > Miscellaneous

For C projects it is shown.

* v2.6.1-201502281154
* Eclipse 4.5.1
* Java 1.7.0_60
* Windows 8.1

ilg-ul, 2015-11-24:
thank you, I'll investigate.

ilg-ul, 2015-11-26:
I checked, and in both cases the **Other linker flags** is visible. See attached images. 

What other box do you mean?

[image]
[image]

joelm-somnium, 2015-11-26:
The Linker flags box is visible in a C project above the Other objects box.

[image]

Attached is a patch that appears to correct the issue.

[issue40-linkerflags-patch.txt](https://github.com/gnuarmeclipse/plug-ins/files/44869/issue40-linkerflags-patch.txt)

ilg-ul, 2015-11-26:
ah, now I see.

merged.

please check the GitHub repository and 2.11.1-201511261230 from updates test.

ilg-ul, 2015-12-17:
fixed in 2.11.1-201512141335

closed 2015-12-17 by ilg-ul