eclipse-embed-cdt/eclipse-plugins #41: Implement an extension point to add custom toolchains

iulia-vasii, 2015-11-25:
Add support for developer users to add new GNU ARM toolchains using an extension point. For example, this could be used to add more versions of an already supported toolchain. See
https://github.com/gnuarmeclipse/plug-ins/issues/33
when a toolchain's prefix was changed in a later version.

iulia-vasii, 2015-11-25:
Attached is the git patch with a fix (I added .txt extension to be able to upload the file here, please remove that before applying it).

[0001-extension-point-to-add-custom-GNU-ARM-toolchains.patch.txt](https://github.com/gnuarmeclipse/plug-ins/files/43739/0001-extension-point-to-add-custom-GNU-ARM-toolchains.patch.txt)

ilg-ul, 2015-11-26:
merged, with some changes.

please check the GitHub repository and 2.11.1-201511261230 from updates test

iulia-vasii, 2015-11-27:
I confirm it works. Thanks!

ilg-ul, 2015-11-27:
Just curious, what toolchain do you plan to add, I already defined all official Linaro ones.

iulia-vasii, 2015-11-27:
At the moment, we need to support both Linaro versions (4.8 & 4.9). On a long run, users should be able to define their own toolchains.

ilg-ul, 2015-11-27:
> both Linaro versions (4.8 & 4.9)

ok, but both have the same prefix/suffix, why do you need different toolchains, just set the path to different folders, possibly per project, at it should work.

iulia-vasii, 2015-11-27:
On bare-metal, 4.8 has the prefix "aarch64-none-elf-", whereas 4.9 "aarch64-elf-".

ilg-ul, 2015-11-27:
ah, ok, I remember. 

but I guess this is just a temporary problem; and for special projects you can always edit the prefix for each project to anything non-standard you need.

iulia-vasii, 2015-11-27:
This is what we do now (without this support), edit the prefix and path when necessary, but having too many manual steps is not desired. And this procedure turned out to be quite error-prone.

Also, this feature would be the starting point when we'll add support for users to define their own custom toolchains (e.g. Yocto SDK toolchains).

ilg-ul, 2015-11-27:
Ok, just be aware that once a toolchain is added, it is expected to be always available; the behaviour of the build process might be unexpected if the toolchain is no longer there.

iulia-vasii, 2015-11-27:
This may be a problem, indeed. Does this apply for all kind of projects, existing an new?

ilg-ul, 2015-11-27:
yes. 

toolchains are identified by name; I don't know what happens if the name stored in .cproject is no longer available, please check (it might be possible that the first toolchain in the array is returned).

iulia-vasii, 2015-11-27:
Yes, I will do some testing. Thanks!

ilg-ul, 2015-12-17:
added to 2.11.1-201512141335

closed 2015-12-17 by ilg-ul

iulia-vasii, 2016-03-21:
In response to your concern mentioned above:

"be aware that once a toolchain is added, it is expected to be always available[...] toolchains are identified by name; I don't know what happens if the name stored in .cproject is no longer available"

I tested and in case the toolchain is not found (by name) the Settings page becomes invalid (the UI is not even depicted). The fix is easy, to return the first toolchain in the array as you suggested. Although this scenario isn't common at all, I'm thinking that user should be notified about this (the toolchain his/her project is configured with was not found and a default selection was made). Where do you think such a warning/info message should be placed:

1. as a global message in the Settings page (upper area), managed by Toolchains tab
2. as a simple message in the Toolchains tab (new label control in the bottom area)
3. validation decorator in the toolchains combo box (Toolchains tab)
4. other ideas?

ilg-ul, 2016-03-21:
I don't know. Could you detect any rule from the behaviour of other plug-ins? If we have multiple choices, I would choose one that preserves UI consistency.

iulia-vasii, 2016-03-28:
I tried to "fix" the Settings page by selecting the default toolchain (first occurrence in list) in case of errors, see #106. I consider this change necessary.

FYI, I managed to notify user by adding an error in Problems view (project marker) without affecting GNU ARM code (satisfactory solution).

ilg-ul, 2016-03-28:
I lost track of this issue. is the extension point functional now?

iulia-vasii, 2016-03-28:
Yes, it is functional.