eclipse-embed-cdt/eclipse-plugins #42: GNU ARM Eclipse Windows Build Tools or ARM plug-ins issue

jtheires, 2015-12-02:
I'm attempting to configure Eclipse to create applications for ARM-based microcontrollers, such as the Nordic nrf51822. According to the instructions at http://embeddedsoftdev.blogspot.com/p/ehal-nrf51.html, I can use Eclipse to do this.

On my Windows 7 Pro laptop, I've installed Eclipse, Java 8, GCC for ARM embedded, ARM Eclipse plugins, and ARM Toolchain, so far. When I get to the build tools (make & rm v2.6), I run into problems. 

After running the setup wizard and successfully checking its version with "make --version" command, I went to check the global tools path as instructed in the directions (http://gnuarmeclipse.github.io/windows-build-tools/install/), but found the "Global Tools Paths" area in Eclipse Preferences windows is not there. The instructions to check if the paths definitions are effective, point to C/C++ Build/Environment - here again, my list of environment variables is blank.

Tried installing an older version of the build tools (2.4) and tried again - same result.

Attached are screen shots of the instructions and my Eclipse installation screens for both the Global Tools Paths issue as well as the environment variables issue.
[Eclipse issues.docx](https://github.com/gnuarmeclipse/plug-ins/files/50306/Eclipse.issues.docx)

ilg-ul, 2015-12-03:
> Tried installing an older version of the build tools (2.4) and tried again - same result.

the problem is not related to the tools, but to the plug-ins.

are you sure the plug-ins installed without problems?  can you check About Eclipse -> Installation Details -> Installed Software, then Plug-ins.  there should be about 22 plug-ins. 

if not, in the plug-ins page, please enter 'ilg' in the filtering field and capture the list of plug-ins. attach it here directly, do not use .doc files, since they are not portable.

you can also check the Eclipse log for exceptions; the logs are in the workspace, the .metadata folder.

jtheires, 2015-12-03:
The Plug-ins section in my Installed Software tab is missing.
Looking in the Plug-ins tab (confusing, BTW) and filtering on ilg yields zero results.
pdf of screen shots attached.
[Eclipse plug-ins.pdf](https://github.com/gnuarmeclipse/plug-ins/files/51196/Eclipse.plug-ins.pdf)

ilg-ul, 2015-12-03:
> Looking in the Plug-ins tab and filtering on ilg yields zero results.

this means you din not install the plug-ins.

jtheires, 2015-12-03:
OK - looking closer at http://gnuarmeclipse.github.io/plugins/features/, one has to ignore all the 'noise' up top and go to the submenu item, cleverly hidden in a side menu, to install the GNU ARM Eclipse Plug-ins. Perhaps the hyperlinks related to installation could point directly to the install pages?
In any event, after installing through the Eclipse Install New Software dialog and restarting Eclipse, the 'missing' Global Tools Paths page has revealed itself to me.
Thanks, Liviu, for your assistance.

ilg-ul, 2015-12-03:
> Perhaps the hyperlinks related to installation could point directly to the install pages?

I don't get it. there is a whole section in the right side menu related to installation. isn't it visible enough?

closed 2015-12-03 by ilg-ul