eclipse-embed-cdt/eclipse-plugins #43: MxCube project

bremenpl, 2015-12-03:
Hello there,
I have been using Arm Gnu Eclipse plugin together with a OpenSTM32 plugin for a while and it worked ok. I generated project in CubeMx and imported it to eclipse using OpenSTM32 plugin. Then built it with the same plugin compiller. I then used Arm Gnu Eclipse only for jlink configuration and debugging.
And today i discovered a huge problem- I cannot see registers view... Both plugins got it covered but it wont work together. I cannot use Packs from Arm Gnu Eclipse, because there is no Device section in the project, as the project is made by OpenSTM32 plugin. Also I cannot see registers from OpenSTM32 plugin, as I am using Jlink not ST-Link... I am stuck.

I was wondering, do you know any workaround for this situation maybe, so I could use Packs with OpenSTM32 project? I would really rather use the Arm Gnu Eclipse project structure but I cannot import project from CubeMX to it... :(
Is there a solution? I would apreciate all help!

ilg-ul, 2015-12-03:
sorry, GNU ARM Eclipse is a standalone set of plug-ins, it is not related to OpenSTM32 and therefore not compatible.

in the future I might add an Import from OpenSTM32, but currently you have to do it manually.

bremenpl, 2015-12-03:
But is there any straight forward way of managing content from cubeMX in a 
gnu arm eclipse project? Lets say i would create an cube mx project and 
then i would have integrate the code somehow?

ilg-ul, 2015-12-03:
>  integrate the code somehow

yes, there are several options, but not particularly convenient, since they require to reconfigure the Eclipse project.

depending on your project, you may opt to generate only the initialisation code in CubeMX and copy/paste it into an existing GNU ARM Eclipse project, then configure the source path, include path and preprocessor symbols.

if you like CubeMX and feel confident with Eclipse, you can also start with an empty project and copy/paste all sources from CubeMX, but there are a lot of details that you have to configure manually.

ilg-ul, 2015-12-03:
you can also ask STM to enhance CubeMX with an Export to GNU ARM Eclipse, which would greatly simplify things.

bremenpl, 2015-12-03:
No harm in asking i guess. Ill do that. Let you know.

bremenpl, 2015-12-03:
Just to have a general view, how long do you think it could take you to 
implement cubeMX import feature and when do you think would you do it?
I would love to participate mysalfe but as i stated before, i dont java at 
all :/.

ilg-ul, 2015-12-03:
> long do you think it could take you to implement cubeMX import

I don't know, it is not on my top priority list, it might take a few months to get to it.

and it'll probably be OpenSTM32 import, i.e. you need to configure CubeMX to generate the OpenSTM32 project, then ask GNU ARM Eclipse to import it.

bremenpl, 2015-12-03:
Ys thats how i though it would work. Thank you for info, ill contact st and 
let you know. But i gues that would also take couple months.

cnoviello, 2015-12-13:
Hi,
I would like to signal this really simple python script I've realized:

https://github.com/cnoviello/CubeMXImporter

It allows to automatically import a project generated with CubeMX inside an Eclipse project generated with the GNU ARM Plugin (which, according to me, is the best option to develop STM32 applications).

Hope it helps.

ilg-ul, 2015-12-13:
Carmino, thank you for the info. 

Perhaps it would be useful to post a more detailed article in the new forum I created at http://www.element14.com/community/groups/gnu-arm-eclipse. Use _Start a discussion_.

cnoviello, 2015-12-13:
Ok, Nice idea. I'll write down something this week.

bremenpl, 2015-12-13:
I am looking forward to it.

cnoviello, 2015-12-14:
Done:

http://www.element14.com/community/thread/48115/l/how-to-quickly-import-a-stm32cubemx-project

ilg-ul, 2015-12-14:
great, thank you.

bremenpl, 2015-12-14:
I will try to port my project tommorow, thank you :).

Ivan-SB, 2016-01-26:
Here is my attempt to import CubeMX projects to ARM GNU plugin
https://github.com/Ivan-SB/lib2eclipse
and it is python 3

Carmine's plugin anyway is much more mature I think.

BTW Liviu I used a trick to be able to remove or update include/libraries etc...
I added an attribute to listOptionValue to know which library added what.
I don't know anything about eclipse internals... so that may be a bad idea in general... but maybe not.
If it makes any sense would you add such a marker to the include paths, define etc... that your plugin is adding?
Do you have a better idea?
Do you know any way from python to generate an id for <option />?

thanks

ilg-ul, 2016-01-26:
> add such a marker

I'm not sure this is possible, the information in .cproject is mostly generated by CDT from the toolchain definitions, and I doubt there is anything like these markers there.

Ivan-SB, 2016-01-26:
Indeed, unfortunately, as soon as you touch anything in the settings, alien attributes get reset.
The other option is to compare all the content to what a library added and delete just the stuff that match. I'll consider to write an "uninstall" file then.

I'll try to learn something about eclipse plugin development and see if I can "scratch my own itch" contributing someway.

thanks