eclipse-embed-cdt/eclipse-plugins #44: SourceTree EHAL clarification needed

jtheires, 2015-12-03:
Continuing to work through Eclipse/ARM installation steps, and got to the point where EHAL for nRF51 needs to be added.
Instructions say "Clone the EHAL source tree.   This can be done using git command line tools or use SourceTree from this site http://www.sourcetreeapp.com.  github : https://github.com/I-SYST/EHAL."
As a newbie to SourceTree and Eclipse (and using Windows 7 platform) I downloaded both the SourceTree app and the EHAL zip file, but I'm unsure where to place the EHAL stuff. Instructions say to mimic a source tree, which is expressed as relative to "/your_root" - where is this "/your_root" folder (by default) on a Win7 Eclipse install? I'm not seeing anything obvious in my folder structure.

ilg-ul, 2015-12-04:
sorry, but how is this question related to GNU ARM Eclipse plug-ins?

jtheires, 2015-12-04:
Not sure. Maybe there's another more suitable discussion?

Regards,
James

jtheires, 2015-12-04:
Where would you suggest my question be posted?

ilg-ul, 2015-12-05:
sorry, I have no experience with nRF51 and EHAL.

closed 2015-12-05 by ilg-ul