eclipse-embed-cdt/eclipse-plugins #45: nRF51 SVD is not being loaded from Nordic DeviceFamilyPack

derekmulcahy, 2015-12-05:
Hi,

I have successfully installed packages for STM32F030 devices and noted that the SVD files were processed. Producing the following in the log.

Parsing SVD file ".../Packages/Keil/STM32F0xx_DFP/1.4.0/SVD/STM32F030.svd"...

When I install the NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0.pack it does not load the nrf51.svd file and the result is that the devices are not available for selection in the Project Properties Device Selection.

I looked through the working pdsc files for STM devices and I noted that STM SVDs are under "subfamily" nodes while the Nordic SVDs are under a  "family" XML node.

Are the Nordic files broken in some way?

OS X 10.11.1
Eclipse 4.5.1
GNU ARM Eclipse plug-ins 201511061603

Regards
Derek

ilg-ul, 2015-12-05:
> Are the Nordic files broken in some way?

I don't know, but I remember some old issues with them.

I'll investigate.

ilg-ul, 2015-12-05:
I installed NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0 and the device selection shows 6 entries:

[image]

the SVDs are not used to populate this window, they are loaded only when referred by the debug plug-in.

derekmulcahy, 2015-12-05:
Good, it seems that it can be made to work!

I shutdown Eclipse, removed the Packages directory and installed just the NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0 package.

My Settings window looks like this.
[image]

The STM32F103RB settings are remnants from the ST Micro devices loaded previously.

Because I was only doing one file I noticed an error in the console complaining about a missing pdsc. Package/.cache contains (amongst others)

NordicSemiconductor.nRF_DeviceFamilyPack.8.0.3.pdsc
NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0.pack

but no NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0.pdsc. I copied/renamed the file into the .cache directory and restarted Eclipse and the Settings->Devices now contains the expected entries!

I am not sure where the 8.0.3.pdsc file came from. I didn't select it.

It is working for me now. Thanks for the plugins and the Saturday support, brilliant.

ilg-ul, 2015-12-05:
there is something strange, the keil index refers to `... version="8.0.3"` and the `NordicSemiconductor.nRF_DeviceFamilyPack.8.0.3.pdsc` seems correct; I have no idea where you got the `... 8.2.0.pack`, there is no such file.

could you help me recheck this? please install a fresh Eclipse, use a fresh workspace and reinstall the same Nordic package, to see if the problem can be reproduced.

derekmulcahy, 2015-12-05:
I moved my old Eclipse and the Packages out of the way. Download and installed eclipse-java-mars-1-macosx-cocoa-x86_64.tar. Created a new workspace. Installed all the gnuarmeclipse plugins. Restarted Eclipse. Refreshed the packs. Installed just the NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0.pack. I then created a new empty C project.

Here are the options that I have for the nRF_DeviceFamilyPack

[image]

I installed the 8.2.0 version. When I go to the Settings -> Devices no devices are available. Again copying the NordicSemiconductor.nRF_DeviceFamilyPack.8.2.0.pdsc into the cache "fixes" it.

I have attached the GNU Arm Eclipse Packs console log.
[console.txt](https://github.com/gnuarmeclipse/plug-ins/files/52926/console.txt)

I tried one more thing, removed Packages, and instead selected the older 8.0.3 DeviceFamily entry and it works fine. nRF51 Devices appear in the Settings.

Cheers,
Derek

ilg-ul, 2015-12-05:
I think I got it, it looks like an out-of-sync problem, the Keil index mentions 8.0.3, but the most recent Nordic PDSC probably reached 8.2.0. I guess my Packs Manager does not cope very well with these cases, I thought the index is authoritative but obviously it is not.

since I plan to add new package types, that will require changes in the Packs Manager, I'll also rewrite the Keil index code.

closed 2017-04-24 by ilg-ul