eclipse-embed-cdt/eclipse-plugins #46: Plugins installation error

clarkleach, 2015-12-05:
I get the following when trying to install Cross Compiler and OpenOCD plugins:

Cannot complete the install because one or more required items could not be found.
  Software being installed: GNU ARM C/C++ Cross Compiler 1.15.2.201511061603 (ilg.gnuarmeclipse.managedbuild.cross.feature.group 1.15.2.201511061603)
  Missing requirement: GNU ARM C/C++ Core 2.10.2.201511061603 (ilg.gnuarmeclipse.core 2.10.2.201511061603) requires 'bundle org.eclipse.debug.core 3.8.0' but it could not be found
  Cannot satisfy dependency:
    From: GNU ARM C/C++ Cross Compiler 1.15.2.201511061603 (ilg.gnuarmeclipse.managedbuild.cross 1.15.2.201511061603)
    To: bundle ilg.gnuarmeclipse.core 0.0.0
  Cannot satisfy dependency:
    From: GNU ARM C/C++ Cross Compiler 1.15.2.201511061603 (ilg.gnuarmeclipse.managedbuild.cross.feature.group 1.15.2.201511061603)
    To: ilg.gnuarmeclipse.managedbuild.cross [1.15.2.201511061603]

ilg-ul, 2015-12-05:
it looks like you do not have CDT. are you sure you installed **Eclipse IDE for C/C++ Developers** and not a different Eclipse?

clarkleach, 2015-12-05:
I thought I did.  I installed it via Synaptics package manager which apprently presents its own set of issues.  I'm in the process of re-installing it manually...

clarkleach, 2015-12-05:
That went much better.  Installed plugins without a hitch.  Now for the real test; see if I can get openocd to work with my FT2232D interface.

ilg-ul, 2015-12-05:
> I installed it via Synaptics package manager

hmmmm... where does the documentation say to use the Synaptics package manager?

when I suggest to follow the documentation 'by the book', I expect people to do exactly so. it is clearly stated that not doing so is a big source of trouble.

closed 2015-12-05 by ilg-ul

clarkleach, 2015-12-05:
Where in the documentation does it say anything about installing Eclipse?

ilg-ul, 2015-12-05:
http://gnuarmeclipse.github.io/plugins/install/#eclipse--cdt

ilg-ul, 2015-12-13:
For such question, in the future please use the GNU ARM Eclipse forum, available at http://www.element14.com/community/groups/gnu-arm-eclipse.