eclipse-embed-cdt/eclipse-plugins #48: Setting watchpoints?

AD7ZJ, 2015-12-18:
Sorry if I missed this somewhere but it appears watchpoints are not supported by the plugin?  I highlight a variable, go to Run->Toggle Watchpoint but it's grayed out.  Has this feature just not been implemented in the plugin yet?  For completeness, here's my setup:

Eclipse: Luna 4.4.1
GNU ARM C/C++ J-Link Debugging: 3.1.1.201412191510	
JLink V4.96
JLink Pro debugger
Windows 7 Pro 64 bit

diabolo38, 2015-12-18:
I didn't knew that way 
To add a watch point go into the variable  window  rigth click on the variable , "add watchpoint "

[image]

to add a watch point on a global select the global in source or add it manulay to the w watch/expression 
view. From the watch view right click , add it  as above. 
I notice that depending on where is the watch view docked the righ click pop up  will not show  thsi possibility if so dock it somewhere else (side to breakpoint)

AD7ZJ, 2015-12-18:
That just shows you the value of the variable, I'm looking for something that will break when the variable gets read, written or both.

diabolo38, 2015-12-18:
have you tried ?
"add watchpoint"  will add an hardware watchpoint  , that is what your looking at  no ?
for global you have first to add it first to the watch view then you can set watchpoint to it

ilg-ul, 2015-12-18:
the support for watchpoints comes from Eclipse CDT, not from the GNU ARM Eclipse plug-ins.

my debugging plug-ins are mainly intended to start the gdb server, to redirect the consoles, to configure everything from the GUI, etc.

AD7ZJ, 2015-12-19:
Thanks diabolo38, I missed that part.  It did seem to work when expressions explicitly wrote to the variable, but did not catch the array overrun that was also writing the variable's address.  That's what I was hoping to use the watchpoint to find.  The JLink eclipse plugin must have something to do with this as the watchpoint address has to make it to the jlink somehow.

diabolo38, 2015-12-19:
In the intermediate  dialog box you can cope with array size (range) and i suspect you can also  change the expression itself  to change something like p  in *p , x, in x[3] to deal with array. 
Note that the JL "chain" must be aware how many and of how to set WP all the wp h/w is on the MCU.

ilg-ul, 2015-12-19:
@diabolo38, could you write a post in the newly created discussion group, to document your findings on watchpoints? (or any other topics you consider useful)

http://www.element14.com/community/groups/gnu-arm-eclipse

closed 2015-12-24 by ilg-ul