eclipse-embed-cdt/eclipse-plugins #49: Run QEMU from Eclipse: Error in final launch sequence

hubmartin, 2016-01-05:
Hi, I successfully compiled ELF file in Eclipse and when I click Debug in GDB QEMU Debugging configuration I get this error:

Error in final launch sequence
Failed to execute MI command:
-exec-run
Error message from debugger back end:
Don't know how to run.  Try "help target".
Don't know how to run.  Try "help target".

Running Eclipse Kepler with 32bit QEMU on Winx64
I tried to run QEMU manually with command line but I don't know which command line arguments I have to pass. I ended up with :

qemu-system-gnuarmeclipse.exe -board NUCLEO-F103RB -mcu STM32F103RB --image ..\images\NUCLEO-F103RB.jpg

Eclipse plugin version:
GNU ARM C/C++ QEMU Debugging	2.1.3.201512141335

Eclipse QEMU console:
GNU ARM Eclipse 32-bits QEMU v2.4.50 (qemu-system-gnuarmeclipse.exe).
Board: 'STM32-H103' (Olimex Header Board for STM32F103RBT6).
Device: 'STM32F103RB' (Cortex-M3 r0p1, MPU), Flash: 128 kB, RAM: 20 kB.
Command line: 'qemu_test 3 2 1' (15 bytes).
Cortex-M3 r0p1 core initialised.
LED: Green active low 14*12 @(97,243) /machine/mcu/stm32/gpio[c],12
GDB Server listening on: 'tcp::1234'...
QEMU 2.4.50 monitor - type 'help' for more information
(qemu) QEMU 2.4.50 monitor - type 'help' for more information
(qemu) Cortex-M3 r0p1 core reset.

I have bit older GCC but I managed to compile blink example to the ELF file.
GDB version:
C:\Program Files (x86)\GNU ARM Eclipse\QEMU\2.4.50-201510290935-dev\bin>arm-none
-eabi-gdb.exe --version
GNU gdb (GDB) 7.2.50.20100908-cvs
Copyright (C) 2010 Free Software Foundation, Inc.
This GDB was configured as "--host=i686-mingw32 --target=arm-none-eabi".

Thank you for any suggestions

ilg-ul, 2016-01-05:
> QEMU ... I don't know which command line arguments

http://gnuarmeclipse.github.io/qemu/options/

the `image` argument defines the application elf file, not the board picture.

> I have bit older GCC

I suggest you use the 4.9.3 version, available from launchpad. (had no time to test the recent 5.2).

so update the toolchain, update the plug-ins and qemu, then do **exactly** the steps in the arm blinky tutorial, using the stm32f4discovery board, and it should work.

once you confirm that your environment is ok, experiment with other boards and configurations.

hubmartin, 2016-01-05:
Thanks, I overlooked that page. With manual command line start of QEMU it works and I can simulate.
Thanks to all that develop this CortexM3 QEMU package, it's awesome.

closed 2016-01-05 by ilg-ul

amalnn, 2021-09-30:
$ qemu-system-gnuarmeclipse.exe -m STM32F4-Discovery -mcu STM32F407VG -kernel finalled.elf
D:\Internship\xpack-qemu-arm-2.8.0-12\bin\qemu-system-gnuarmeclipse.exe: -m STM32F4-Discovery: Parameter 'size' expect
s a size

How to fix this

TommyMurphyTM1234, 2021-09-30:
> $ qemu-system-gnuarmeclipse.exe -m STM32F4-Discovery -mcu STM32F407VG -kernel finalled.elf D:\Internship\xpack-qemu-arm-2.8.0-12\bin\qemu-system-gnuarmeclipse.exe: -m STM32F4-Discovery: Parameter 'size' expect s a size
> 
> How to fix this

Pass an appropriate -size option?

https://bugs.launchpad.net/qemu/+bug/1707297

ilg-ul, 2021-09-30:
Please read the manual page on how to invoke QEMU manually:

- https://xpack.github.io/qemu-arm/options/

ilg-ul, 2021-09-30:
For questions and support please use the project forums.