eclipse-embed-cdt/eclipse-plugins #50: Create plugin or use pack?

buzmeg, 2016-01-08:
I'd like to create something like the templates for STMxxxxx, but for the Nordic chips instead.

What is the correct method of doing this?  Templates?  Or use the new packs?  I'd rather not go through a lot of useless details for templates only to have it all obviated by packs in the next couple of months.  I would rather learn the new thing and deal with some instability rather than learn a bunch of details that are a dead end.

I have managed to pull and compile everything with maven.  I even created a project to encompass what I need for the Nordic chips, but then I run into having no idea how to integrate this into the Maven build.  There is lots of Maven magic in the .parent project and it is not clear how to extend that.

Sorry for creating an issue on this, but it's not clear where else I should make this query.  If there is a better place/forum, please give me a pointer and I will go there.

Thanks.

ilg-ul, 2016-01-08:
> Nordic chips

did Nordic provide now freely all libraries? because I know of another Nordic plug-in, but is requires to separately install the non-public libraries.

regarding your question, I'm currently working on a solution to migrate the existing templates to packages, so, from my point of view, the templates are dead, I will no longer invest any efforts in maintaining them once the new packs based wizard will be operational.

but this might take a few more months, so if you are in a hurry, and the template is already functional, you can use it.

maven is complicated, probably for a single plug-in you can do the builds with Eclipse.

> ... forum

the forum link is

http://www.element14.com/community/groups/gnu-arm-eclipse

but your request is a feature request, and it is ok to post it here.

buzmeg, 2016-01-08:
> did Nordic provide now freely all libraries?

I don't know.  I haven't reached the point where I was worrying about that yet.  It doesn't seem like I need to log in to download the SDK anymore and I see some BSD-like licenses on things.

> I know of another Nordic plug-in

The one I found got abandoned by the author and no longer works with the latest Eclipse/ARM stuff.  The author also never provided source code to allow others to compile it.  If you know of one that is better than that, I'm all ears.

> but this might take a few more months, so if you are in a hurry

Not in a hurry, but there has been very little progress over the past year at making the Nordic chips work better with Eclipse from the Nordic side (ie. actually making managed build work--Makefile build somewhat defeats the point of using Eclipse).  I suspect it's simply that Nordic doesn't have the resources to support anything other than Keil directly.  If there is progress, then I can swallow hard and justify jumping through hoops to my managers until the new stuff hits.  

I do see from this issue:
https://github.com/gnuarmeclipse/plug-ins/issues/45

that you are apparently able to at least somewhat read the "pack" that Nordic itself prepares for Keil.  Obviously, me helping to debug this would be a far more ideal solution than me bumbling around all by myself.  Embedded stuff already suffers from too little manpower; I'd much prefer to throw in with what the important people are already doing and help out.

Thanks for taking the time to reply.

ilg-ul, 2016-01-08:
> somewhat read the "pack" that Nordic itself prepares for Keil

yes, my Packs Manager reads all Keil packs.

the plan for the future wizard is to create empty CMSIS projects for all processors having a CMSIS pack, so Nordic should be covered. 

for anything more than an empty project... it is hard to tell, since generally the Keil packs provide little support for GCC.

closed 2016-01-25 by ilg-ul