eclipse-embed-cdt/eclipse-plugins #54: semihosting output kills interrupts

curtisembedded, 2016-01-19:
Hello,
This has been a long standing problem for me so I purchased a Segger J-Link-Pro debugger. (as you suggested) The original thought was that this was an OpenOCD issue.

The debug output is much faster with the Segger but it is still killing interrupts and causing the application to fail. I am running under FreeRTOS so I tried to create an Idle task to output a single character at a time by modifying _write(). Unfortunately this is not used with SEMIHOSTING.

How can I fix this problem?

TIA

ilg-ul, 2016-01-20:
> still killing interrupts

what do you mean? by definition using semihosting makes the application non real time, since the processor is halted during calls to the debugger.

> How can I fix this problem?

do not use semihosting. create a new test project, be sure you do not select semihosting syscalls, and for trace select ITM (not available on smaller cores). you then need to configure J-Link to use SWD, and compute the SWO frequency as a divisor of your system clock.

trace_printf() uses trace_write() as implementation, not _write().

ilg-ul, 2016-01-20:
for general question (not bugs or enhancements), please use the project forum:

http://www.element14.com/community/groups/gnu-arm-eclipse

closed 2016-01-25 by ilg-ul