eclipse-embed-cdt/eclipse-plugins #55: different include paths for Release and Debug (relative vs absolute + &quot;)

Ivan-SB, 2016-01-21:
Hi,

I was writing a python tool to gather information from the .cproject file and automatically add some external libraries.

I noticed that the path to the include is a bit different between Release and Debug in templates.
In Debug it is an absolute path without &quot; surrounding it. In Release it seems a relative path surrounded by &quot;
I'm not sure if it is a bug, but I've the feeling it is not what you were planning to write.

Files involved are:
./ilg.gnuarmeclipse.templates.stm/templates/stm32.*/template.xml

thanks

ilg-ul, 2016-01-21:
thank you, I'll investigate.

ilg-ul, 2016-01-21:
I checked the stm32f4 template.xml and could not identify any difference between Release and Debug related to include paths, all are relative and surrounded by quotes.

I suggest you create a new project and run your analyser on it, your .cproject might be corrupted.

Ivan-SB, 2016-01-21:
You're right.

I saw some differences in how paths were specified in the templates[1] and found the same differences in the project and was too quick to think it was a problem in the template.
Apparently at a certain point eclipse decided to convert some relative path to absolute.
Something I would remember if I did it since include path are "too many" to change them inadvertently.

BTW I'm trying to write a CubeMX -> GNU ARM plugin importer or possibly, more generally a library importer/rebaser for GNU ARM plugin. Unfortunately my Java-fu is too weak to do it directly in eclipse.

Sorry for the fuss and thanks for the plugin

[1]
<simple name="path" value="$(sysDir)/$(includeDir)/$(cmsisDir)" />
vs
<element value="&quot;../$(sysDir)/$(includeDir)&quot;" />

closed 2016-01-25 by ilg-ul

Ivan-SB, 2016-01-25:
Did you find the problem or was I just hallucinating?

I'm sorry I haven't been able to reproduce the problem yet, but considering the includes are several I really exclude I changed them, I would have remembered it, maybe something got triggered when examining settings that made eclipse resolve the relative path to absolute and trim the &quot;

thanks

ilg-ul, 2016-01-25:
I'm not sure, but it looks like renaming header files trigger a broken refactoring process that corrupts the .cproject. 

my plug-ins do not deal with these issues, I guess it must be a CDT problem.