eclipse-embed-cdt/eclipse-plugins #56: "STM32F0xx C/C++ Project" template is broken (again)

bjornfor, 2016-01-22:
Originally reported at http://sourceforge.net/p/gnuarmeclipse/bugs/157/, this regression has happened again. Now testing with latest plug-in version, v2.11.1 (201501191402):

1. Create new project, select the "STM32F0xx C/C++ Project" template and complete the wizard using default settings
2. (Try to) build project

Build error:

```
Building file: ../system/src/cortexm/exception_handlers.c
Invoking: Cross ARM C Compiler
arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra  -g3 -DDEBUG -DUSE_FULL_ASSERT -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSTM32F051 -DUSE_STDPERIPH_DRIVER -DHSE_VALUE=8000000 -I"../include" -I"../system/include" -I"../system/include/cmsis" -I"../system/include/stm32f0-stdperiph" -std=gnu11 -MMD -MP -MF"system/src/cortexm/exception_handlers.d" -MT"system/src/cortexm/exception_handlers.o" -c -o "system/src/cortexm/exception_handlers.o" "../system/src/cortexm/exception_handlers.c"
../system/src/cortexm/exception_handlers.c: In function 'dumpExceptionStack':
../system/src/cortexm/exception_handlers.c:91:39: error: 'SCB_Type' has no member named 'HFSR'
   trace_printf (" HFSR =  %08X\n", SCB->HFSR);
                                       ^
../system/src/cortexm/exception_handlers.c:92:39: error: 'SCB_Type' has no member named 'DFSR'
   trace_printf (" DFSR =  %08X\n", SCB->DFSR);
                                       ^
../system/src/cortexm/exception_handlers.c:93:39: error: 'SCB_Type' has no member named 'AFSR'
   trace_printf (" AFSR =  %08X\n", SCB->AFSR);
                                       ^
../system/src/cortexm/exception_handlers.c: In function 'HardFault_Handler_C':
../system/src/cortexm/exception_handlers.c:314:23: error: 'SCB_Type' has no member named 'MMFAR'
   uint32_t mmfar = SCB->MMFAR; // MemManage Fault Address
                       ^
../system/src/cortexm/exception_handlers.c:315:22: error: 'SCB_Type' has no member named 'BFAR'
   uint32_t bfar = SCB->BFAR; // Bus Fault Address
                      ^
../system/src/cortexm/exception_handlers.c:316:22: error: 'SCB_Type' has no member named 'CFSR'
   uint32_t cfsr = SCB->CFSR; // Configurable Fault Status Registers
                      ^
../system/src/cortexm/exception_handlers.c:326:12: error: 'SCB_Type' has no member named 'DFSR'
   if (((SCB->DFSR & SCB_DFSR_BKPT_Msk) != 0)
            ^
../system/src/cortexm/exception_handlers.c:326:21: error: 'SCB_DFSR_BKPT_Msk' undeclared (first use in this function)
   if (((SCB->DFSR & SCB_DFSR_BKPT_Msk) != 0)
                     ^
../system/src/cortexm/exception_handlers.c:326:21: note: each undeclared identifier is reported only once for each function it appears in
../system/src/cortexm/exception_handlers.c:327:15: error: 'SCB_Type' has no member named 'HFSR'
       && ((SCB->HFSR & SCB_HFSR_DEBUGEVT_Msk) != 0))
               ^
../system/src/cortexm/exception_handlers.c:327:24: error: 'SCB_HFSR_DEBUGEVT_Msk' undeclared (first use in this function)
       && ((SCB->HFSR & SCB_HFSR_DEBUGEVT_Msk) != 0))
                        ^
../system/src/cortexm/exception_handlers.c:332:14: error: 'SCB_Type' has no member named 'HFSR'
           SCB->HFSR = SCB_HFSR_DEBUGEVT_Msk;
              ^
system/src/cortexm/subdir.mk:24: recipe for target 'system/src/cortexm/exception_handlers.o' failed
make: *** [system/src/cortexm/exception_handlers.o] Error 1

20:36:37 Build Finished (took 987ms)
```

Eclipse version: 4.3 (Kepler SR2)
Java: openjdk version "1.8.0_60"
OS: Linux 64-bit (NixOS)
Toolchain version: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]

ilg-ul, 2016-01-22:
looks like a wrong update somewhere, I'll investigate.

bjornfor, 2016-01-25:
Sorry, my mistake. I was running old plugin version after all (https://github.com/gnuarmeclipse/plug-ins/issues/57#issuecomment-174667808). There is no problem with plugin v2.11.1.

closed 2016-01-25 by bjornfor