eclipse-embed-cdt/eclipse-plugins #57: Packs manager download logic is brittle

bjornfor, 2016-01-22:
Plug-in version: v2.11.1 (201501191402)
Eclipse version: 4.3 (Kepler SR2)
Java: openjdk version "1.8.0_60"
OS: Linux 64-bit (NixOS)

I had difficulties with getting any "packs" installed. Clicking the refresh button in the Packs perspective would start downloading stuff, but in the end no packs would show up (after the process completed after a couple of minutes). There were errors about missing/broken URLs on the console (eclipse stdout/stderr).

Even though some links were broken, not all were. In the end I succeeded installing some packs by manually downloading http://www.keil.com/pack/index.idx (the default packs URL) and removing everything I knew I didn't need. Then going back to Eclipse, press the refresh button, a list of packs would show up and I could install them.

I have a theory that one or more broken downloads causes the gnuarmeclipse packs manager downloader to give up on the working downloads too. That should be improved. I think it should show the packages it managed to download, even if some failed.

ilg-ul, 2016-01-22:
you are rightpacks manager has problems to cope with the unusual situations that occur with some  unavailable servers.

for the moment just click the Ignore button during the Refresh and it should continue to read the index.

it is planned to accommodate a new type of packages, so the packs manager will be reworked, but I don't have a date when ready.

bjornfor, 2016-01-23:
I cannot find the "Ignore" button. (I'm in the Packs perspective.) Where is it?

ilg-ul, 2016-01-23:
when an error occurs during Refresh, a message box is displayed and you can Cancel, Ignore, or Retry.

bjornfor, 2016-01-23:
There is no message box on my system :-(

Could it be that my Eclipse is too old (Kepler, 4.3 SR2)?

FYI, I see three tabs with info in the bottom of Eclipse:

1. Progress. A progress bar steadily progressing :-)
2. Console. 
```
[...]
Processing "/home/bfo/Packages/.cache/YOGITECH.fRSTL_ARMCMx_EVAL.1.0.1.pdsc" for content.xml ...
Schema version "1.2".
Completed in 209ms.
Processing "/home/bfo/Packages/.cache/YOGITECH.fRSTL_STM32Fx_EVAL.1.0.0.pdsc" for content.xml ...
Schema version "1.2".
Completed in 109ms.
Processing "/home/bfo/Packages/.cache/lwIP.lwIP.1.4.1.pdsc" for content.xml ...
Schema version "1.2".
Completed in 838ms.
Failed with "Failed to open connection, response code 403", ignored
Copy 517B
 from "http://gnuarmeclipse.sourceforge.net/packages/content.xml"
 to   "/home/bfo/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"

Update packs completed in 183s.
2016-01-24 00:33:37
Loading repos summaries...
Parsing cached content file "/home/bfo/Packages/.cache/.content_www_keil_com_pack.xml"...
File does not exist, ignored.
Parsing cached content file "/home/bfo/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"...
File parsed in 1ms.
Identifying installed packages...
Found no installed packages.
Completed in 1ms.

2016-01-24 00:33:37
Collecting devices...
Found none.
Completed in 1ms.

2016-01-24 00:33:37
Collecting packs...
Found none.
Completed in 1ms.
```
3. Problems. Shows a warning from Packs ```Failed with "Failed to open connection, response code 403", ignored```. (And an informational note that the refresh operation completed.)

ilg-ul, 2016-01-23:
please install a separate instance of the latest Eclipse and the latest plug-ins and try again.

bjornfor, 2016-01-24:
Ah, in Eclipse 4.4[note1] the whole downloading process is shown in a modal window. The window with the "Ignore" button opened in a new modal window _perfectly placed behind_ the (now) frozen progress window. I took a while to find it, for the longest time I just thought the downloader was hung :-)

Now I have all packs that were downloadable and didn't have to manually edit the index file. Much better :-)

[note1]: I tried updating to Eclipse 4.5, but something has changed in how Eclipse finds its plugins, so the way the Nix package manager installs the gnuarmeclipse plugin does not result in Eclipse finding it :-(

ilg-ul, 2016-01-24:
the GNU ARM Eclipse install docs clearly state to install Eclipse directly from eclipse.com, and avoid the distribution package manager.

bjornfor, 2016-01-25:
The docs say not to use a package manager because they often distribute old Eclipse versions. But they also say that Eclipse 4.3, the version I reported running, _is_ supported. I'll update the issue title.

(Btw, the Nix package manager is source based. I can use it to install any Eclipse version from eclipse.org. It is a very different beast than apt :-) )

ilg-ul, 2016-01-25:
>  Nix package manager is source based. I can use it to install any Eclipse version ... I tried updating to Eclipse 4.5 ... gnuarmeclipse plugin does not result in Eclipse finding it  ... I'll update the issue title.

wrong, the Eclipse version might have its share for the blame, but if you want to stay away from trouble, you should use the latest versions (Eclipse + plug-ins). if your distribution package manager was not able to install a functional Eclipse Mars.1 with all required plug-ins, it is a good indication that you should install Eclipse manually, as suggested.

also I could not identify any plug-in released in Jan 19, 2015; v2.11.1 was released in Dec. 2015. are you sure you initially used the latest version of the plug-ins?

install Eclipse and the plug-ins **exactly** as described in the docs and it should work.

diabolo38, 2016-01-25:
Just to do a quick comment on  fact to always update to mars or latest eclipse is not always possible. 
For instance i have some "old" 3.4x86-32 linux and 10.6.8 mac that cannot (easily) have java upgrade hence will not run latest eclipse (anything after kepler SR2 actually). 
I think the big c/c++ dev part of the plugin still work on this system but at some point it could brake :( mean i will get stuck with my old version or coudl heven brake it trying to upgrade.

I will not push toward keeping old suff compatibility i know how painful it is  but keeping some consideration for it will be welcome

ilg-ul, 2016-01-25:
I just downloaded an old Kepler 4.3 SR2 from eclipse.org and installed it on my Mac; then I installed the latest official plug-ins and clicked Refresh.

there were 2-3 packages not available, but the modal dialog was displayed correctly, on top of the progress bar, and clicking Ignore was not a problem.

I guess that if you repeat these steps, you will reach the same conclusion on your Linux too.

so I seriously doubt that your claims about plug-ins misbehaving on some Eclipse versions are real.

I'd rather suspect a misconfigured Eclipse, or using older plug-ins.

bjornfor, 2016-01-25:
Yes, I think you're right about older plugin. The version number I reported looks odd. I'll test when I get home from work.

ilg-ul, 2016-01-25:
there's no mystery here, if you install the plug-ins from the update site (or Eclipse Marketplace), you get the most recent version; if you use alternate solution, or fancy installers, most probably you end up with older versions.

don't you remember how you installed the plug-ins in your original setup?

bjornfor, 2016-01-25:
I found the issue. It was a user mistake.

What I did was installing the latest gnuarmeclipse plugin (v2.11.1) using "offline method", by extracting the zip file into the "dropins" folder (all automated). But I also had, unknowingly, an old install of the plugin in ~/.eclipse, from when I used the "update site" method (probably a year ago or so). And apparently the plugin in ~/.eclipse takes precedence over "dropins". So there I was thinking I was running v2.11.1 when I really was not.

And here is why I reported conflicting plugin version and timestamp at the top. For some reason (possibly to verify the install?), instead of just using the timestamp in the zip file (filename) that I had downloaded, I went into Eclipse to view installation details. And there was the timestamp from the _old_ plugin. And I didn't notice :-(

@ilg-ul: Sorry for wasting your time! Next time I'll try to remember ```mv ~/.eclipse ~/.eclipse.bak```.

bjornfor, 2016-01-25:
(The modal window with "Ignore" button still shows up _right_ behind the progress bar window here. But oh well. Now I know.)

closed 2016-01-25 by bjornfor

ilg-ul, 2016-01-25:
> Next time I'll try to remember mv ~/.eclipse ~/.eclipse.bak.

what about next time to forget about manual install and just use the Eclipse Marketplace or the automated update site, as everyone else?

bjornfor, 2016-01-25:
@ilg-ul: Because my understanding is that those methods only allow for installing the _latest_ version. Maybe that is a misunderstanding? I didn't dig deep.

Anyway, assuming the "only latest version" install... That is often what you want, but not _always_. Maybe you want to setup an environment for an old project (and you don't want to "port" it to the newest version yet)? Maybe you want to work on two different projects on one PC, and those projects require different plugins / Eclipses?

Using a package manager / automation tool solves these issues for me. How do you solve such issues?

bjornfor, 2016-01-25:
Oh, Eclipse Mars.1 works now too. The problem with this new Eclipse was just due to the old plugin in ~/.eclipse. So this was all PEBCAK...

ilg-ul, 2016-01-25:
> those projects require different plugins / Eclipses?

For each project I have a folder with:

- a folder with one or more Eclipses (all installed from original eclipse.org archives)
- a folder with one or many Eclipse workspaces, 
- multiple folders with git repositories.

As for installing different plug-ins, I download the archives and then use them as source for **Install New Software...** (you can enter an url, a folder or an archive)