eclipse-embed-cdt/eclipse-plugins #58: Undefined reference to `_getpid', to `_kill', and others

guarndt, 2016-01-27:
Dear Liviu,

I keep bumping into the same problem over and over again when trying to build projects for the [Ublox C027](https://developer.mbed.org/platforms/u-blox-C027/) with a Cortex M3 NXP LPC1768 CPU.

My development platform:
* Windows 7 x64
* Eclipse Luna 4.4.2 with your (otherwise very helpful) up-to-date GNU ARM Eclipse plugins
* GNU ARM Eclipse build tools 2.6-201507152002
* GNU Tools ARM Embedded 5.2 2015q4

To simplify reproducing the problem, I have attached the required [Eclipse projects](https://github.com/gnuarmeclipse/plug-ins/files/106533/guarndt_c027_mqtt.zip) except for the [mbed library](https://github.com/mbedmicro/mbed) built as described [here](https://developer.mbed.org/teams/SDK-Development/wiki/Mbed-SDK-build-script-introduction); it should be added to the workspace as a project named `mbed' for the references in the given projects to match.

When trying to build the project C027_MQTT, I get the following error message:
```
13:04:05 **** Incremental Build of configuration Debug for project C027_MQTT ****
make all 
Building target: C027_MQTT.elf
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -mthumb-interwork -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -ffreestanding -T "C:\firedect\git\mbed\build\mbed\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM\LPC1768.ld" -Xlinker --gc-sections -L"C:\firedect\dev\workspace\MQTTPacket" -L"C:\firedect\dev\workspace\C027_Support" -L"C:\firedect\git\mbed\build\rtos\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM" -L"C:\firedect\git\mbed\build\mbed\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM" -Wl,-Map,"C027_MQTT.map" --specs=nano.specs -u _printf_float -u _scanf_float -v -o "C027_MQTT.elf"  ./main.o  C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/board.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/cmsis_nvic.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/mbed_overrides.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/retarget.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/startup_LPC17xx.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/system_LPC17xx.o -lMQTTPacket -lC027_Support -lrtx -lrtos -lmbed -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys
Using built-in specs.
Reading specs from c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/nano.specs
rename spec link to nano_link
rename spec link_gcc_c_sequence to nano_link_gcc_c_sequence
rename spec cpp to nano_cpp
COLLECT_GCC=arm-none-eabi-g++
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ tools\ arm\ embedded/5.2\ 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /home/build/work/GCC-5-0-build/src/gcc/configure --build=i686-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/home/build/work/GCC-5-0-build/install-mingw --libexecdir=/home/build/work/GCC-5-0-build/install-mingw/lib --infodir=/home/build/work/GCC-5-0-build/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/home/build/work/GCC-5-0-build/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-5-0-build/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-5-0-build/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build/work/GCC-5-0-build/install-mingw/arm-none-eabi --with-libiconv-prefix=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-gmp=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-mpfr=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-mpc=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-isl=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-cloog=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-libelf=/home/build/work/GCC-5-0-build/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r,armv8-m.base,armv8-m.main
Thread model: single
gcc version 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848] (GNU Tools for ARM Embedded Processors) 
COMPILER_PATH=c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/
LIBRARY_PATH=c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/armv7-m/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/armv7-m/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../arm-none-eabi/lib/armv7-m/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/;c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../arm-none-eabi/lib/
COLLECT_GCC_OPTIONS='-mcpu=cortex-m3' '-mthumb' '-mthumb-interwork' '-Os' '-fmessage-length=0' '-fsigned-char' '-ffunction-sections' '-fdata-sections' '-ffreestanding' '-T' 'C:\firedect\git\mbed\build\mbed\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM\LPC1768.ld' '-LC:\firedect\dev\workspace\MQTTPacket' '-LC:\firedect\dev\workspace\C027_Support' '-LC:\firedect\git\mbed\build\rtos\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM' '-LC:\firedect\git\mbed\build\mbed\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM' '-specs=nano.specs' '-u' '_printf_float' '-u' '_scanf_float' '-v' '-o' 'C027_MQTT.elf'
 c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/collect2.exe -plugin c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/liblto_plugin-0.dll -plugin-opt=c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/lto-wrapper.exe -plugin-opt=-fresolution=C:\Users\guarndt\AppData\Local\Temp\ccLAWjIC.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc_nano -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc_nano --sysroot=c:\program files (x86)\gnu tools arm embedded\5.2 2015q4\bin\../arm-none-eabi -X -o C027_MQTT.elf -u _printf_float -u _scanf_float c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/armv7-m/crti.o c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/armv7-m/crtbegin.o c:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/armv7-m/crt0.o -LC:\firedect\dev\workspace\MQTTPacket -LC:\firedect\dev\workspace\C027_Support -LC:\firedect\git\mbed\build\rtos\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM -LC:\firedect\git\mbed\build\mbed\TARGET_UBLOX_C027\TOOLCHAIN_GCC_ARM -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/armv7-m -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/armv7-m -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../arm-none-eabi/lib/armv7-m -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1 -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib -Lc:/program files (x86)/gnu tools arm embedded/5.2 2015q4/bin/../arm-none-eabi/lib --gc-sections -Map C027_MQTT.map ./main.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/board.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/cmsis_nvic.o C:/firedect/git/mbed/build/mbed/TARGET_UBLOX_C027/TOOLCHAIN_GCC_ARM/mbed_overrides.o C:/firedect/git/mbed/build/mbed/TARGET_U [... cut ...] etpid-and-sbrk). Those did not seem helpful to me, as they often suggest workarounds in code, which are obviously not required: with a common Makefile, it works, so it is probably just a configuration issue in the project, maybe library inclusion order or something? I've switched many options in Eclipse's project config, like newlib-nano, which did not really change anything.

It would be great to hear if you can help here. Honestly, I'm not even sure if this is a bug or a configuration issue...

guarndt, 2016-01-27:
A short update: the project compiles now, it seems to depend on the order of library inclusion somehow...

The screenshot shows that I added libc and libstd++ twice, which makes it work.
I know that the order of inclusion matters, but I do not understand how adding them *first* helps, as they should not introduce new dependencies?

[image]

ilg-ul, 2016-01-27:
the link errors you mentioned are common for newlib, they expect the application to implement some posix functions. linking 'nosys' provides empty content for all missing functions.

please document your findings in the project forum:

http://www.element14.com/community/groups/gnu-arm-eclipse

guarndt, 2016-01-28:
Thank you for the quick reply. It's good to know about the dependency you've explained, but I can't fully agree: As you can see from the above screenshot, nosys is linked already, but that's not enough - I had to link c and stdc++ twice. However, as it seems to work now, I won't bother you further.

closed 2016-01-28 by guarndt

ilg-ul, 2016-01-28:
The GCC linker is a bit picky, when you specify libraries on the command line, the order is important, since it processes the list in a single pass, and if a library refers to a previously processed library, these symbols will not be resolved.

It might be possible to enter libraries in the linker script, and there force the linker to reprocess the list until all references are solved, but I don't know the details, you must check the linker manual.

JojoS62, 2017-07-23:
old issue, but I'm trying to get the lpcxpresso exporter working. There I ran also into this problem. Adding -lnosys did not help, the solution was now to add the GROUP command as the first in the linker script. 

```
GROUP (
  libnosys.a
)
```