eclipse-embed-cdt/eclipse-plugins #59: Unable to use suspend button while debugging, have to telnet into openocd instead.

dholth, 2016-01-27:
I'm using the OpenOCD plugin and the latest https://launchpad.net/gcc-arm-embedded to debug a STM32 project, but the suspend button in the IDE doesn't work. Instead, I have to telnet localhost 4444 and send 'halt' directly to OpenOCD. (After ending a debug session, I also seem to have to unplug and re-plug the USB debugger before I can use it a second time). Am I missing something? Thanks.

gcc-arm-none-eabi-5_2-2015q4
OpenOCD/0.10.0-201601101000-dev/bin/openocd
GNU ARM C/C++ OpenOCD Debugging	3.2.1.201512141335
Eclipse Platform	4.5.0.v20150603-2358
OSX 10.11.2

ilg-ul, 2016-02-09:
sorry, I missed your issue. is it still present?

dholth, 2016-02-09:
Yes, I'm still having this problem on OSX. I haven't had a chance to test on other platforms.

ilg-ul, 2016-02-09:
could you also test with another board, using a different configuration script?

in previous tests on my stm32f4discovery I did not encounter this problem.

dholth, 2016-02-09:
I'll give it a shot.

Links:

  1. https://github.com/gnuarmeclipse/plug-ins/issues/59#issuecomment-181915736

ilg-ul, 2016-02-09:
please do not reply by email, use the web interface.

a-v-s, 2017-03-20:
I am experiencing this problem as well.
Debugger interface: st-link v2
Device: stm32f103c8t6

I am running ArchLinux, 
Eclipse Version: Neon.2 Release (4.6.2) Build id: 20161208-0600
GNU ARM C/C++ OpenOCD Debugging	4.1.3.201702251311
OpenOCD version 0.10.0

ilg-ul, 2017-03-20:
@a-v-s, if you are using gdb 7.12, this is a known Neon.2 bug. use GDB 7.10 until Neon.3 is out.

a-v-s, 2017-03-20:
I am using version GNU gdb (GDB) 7.12.1 (from the archlinux repo) 
I'm trying gdb from linaro 4.9-2016.02. This seems to solve the problem.

closed 2017-04-24 by ilg-ul