eclipse-embed-cdt/eclipse-plugins #61: Offline CMSIS pack install

mayl, 2016-02-05:
Hi,

I've been trying to find a way to install CMSIS packs offline.  The only thing I could find which referenced doing this was an old sourceforge issue: http://sourceforge.net/p/gnuarmeclipse/feature-requests/98/

Can you describe the best way to install a CMSIS pack without connection to the internet?  How do I create the correct index.html file?  Is there a template somewhere I can follow?

Thanks for your help!

ilg-ul, 2016-02-05:
sourceforge.net is dow now, and I cannot access the link, but, if I remember right, there were some references on how to create a local install site.

first, this is method is not supported, and I did not try it myself, so you're on thin ice.

what might probably work, is to install a local web site, including an index.idx as the one used on the keil site, and point the packs manager to it.

there is a post in the project forum on how to add a new packs site, maybe it can help too.

for future questions, please use the project forum.

http://gnuarmeclipse.github.io/support/forum/

mayl, 2016-02-05:
Thank you for pointing me in the right direction.  I'll put any follow on questions in the forum.

closed 2016-02-09 by ilg-ul