eclipse-embed-cdt/eclipse-plugins #64: Pack relative paths

in4lio, 2016-02-10:
Please add a ${variable} that will point to the folder of selected CMSIS Pack for the current project.
Something like ${ProjDirPath}.

ilg-ul, 2016-02-10:
ok, to remain consistent with the other definitions, the first suggestion for the variable name is `packs_path`.

in4lio, 2016-02-11:
Yes, `packs_path` will point to `Packages` folder.

It would be nice to have variables for all installed devices, like Keil 5 has `$$Device:NAME$` variables, e.g. `$$Device:STM32F103$`points to `c:\Keil5\ARM\Pack\STM\STM32F1xx\2.0.1\`. So we will be able to use `${packs_path}/${STM32F103}`.

Also `${packs_path}/${selected_device}` for the current project, could be very useful, I think)

in4lio, 2016-02-11:
A small correction, `$$Device:NAME$` variables in Keil also depend on the selected package.

ilg-ul, 2016-02-11:
I don't think that more package specific variables are realistic, there are too many.

anyway, package management should be automatic, a wizard should create all paths relative to `${packs_path}`, so there should be no need to manually manage so many paths.

I am planning such a new wizard, but it'll take a while to implement it.

in4lio, 2016-02-11:
This way also seems to be quite reasonable.