eclipse-embed-cdt/eclipse-plugins #65: Error creating STM32F10x C/C++ project (on 2.12.1)

in4lio, 2016-02-10:
__plug-in__  2.12.1
Eclipse Mars.1 Release (4.5.1) 20150924-1200
Java 8 Update 73
Windows 10
toolchain 4.9 2015q3

I try to create new project for STM32F10x:
```
Eclipse Menu → File → New → C Project → STM32F10x C/C++ Project → Cross ARM GCC
All settings by default → Finish
```
the next error dialog appears:
```
Error Creating Project

Project connot be created
Reason:
Internal Error:
Details >> java.lang.NullPointerException
```
I have checked a couple of other STM32 projects - the same error...
Hello World C/C++ Project has been created without problem.

Report:
```
The following data will be sent:

------
STATUS
------
pluginId            org.eclipse.cdt.ui
pluginVersion       5.11.0.201509131935
code                4
severity            4
message             HIDDEN
fingerprint         4cadaa19

Exception:java.lang.NullPointerException: HIDDEN
	 at HIDDEN.HIDDEN(HIDDEN:-1)
	 at org.eclipse.cdt.core.templateengine.process.Process.process(Process.java:163)
	 at org.eclipse.cdt.core.templateengine.process.ConditionalProcessGroup.process(ConditionalProcessGroup.java:219)
	 at org.eclipse.cdt.core.templateengine.process.TemplateProcessHandler.processAll(TemplateProcessHandler.java:65)
	 at org.eclipse.cdt.ui.templateengine.Template$1.execute(Template.java:192)
	 at org.eclipse.ui.actions.WorkspaceModifyOperation$1.run(WorkspaceModifyOperation.java:108)
	 at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2241)
	 at org.eclipse.ui.actions.WorkspaceModifyOperation.run(WorkspaceModifyOperation.java:130)
	 at org.eclipse.cdt.ui.templateengine.Template.executeTemplateProcesses(Template.java:202)
	 at org.eclipse.cdt.managedbuilder.ui.wizards.MBSWizardHandler.doTemplatesPostProcess(MBSWizardHandler.java:639)
	 at org.eclipse.cdt.managedbuilder.ui.wizards.MBSWizardHandler.createProject(MBSWizardHandler.java:559)
	 at org.eclipse.cdt.ui.wizards.CDTCommonProjectWizard$1$1$1.run(CDTCommonProjectWizard.java:259)
	 at org.eclipse.ui.actions.WorkspaceModifyDelegatingOperation.execute(WorkspaceModifyDelegatingOperation.java:70)
	 at org.eclipse.ui.actions.WorkspaceModifyOperation$1.run(WorkspaceModifyOperation.java:108)
	 at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2241)
	 at org.eclipse.ui.actions.WorkspaceModifyOperation.run(WorkspaceModifyOperation.java:130)
	 at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:463)
	 at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:371)
	 at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:1002)
	 at org.eclipse.cdt.ui.wizards.CDTCommonProjectWizard$1$1.run(CDTCommonProjectWizard.java:268)
	 at org.eclipse.ui.internal.UILockListener.doPendingWork(UILockListener.java:162)
	 at org.eclipse.ui.internal.UISynchronizer$3.run(UISynchronizer.java:154)
	 at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:35)
	 at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:135)
	 at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4155)
	 at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3772)
	 at org.eclipse.jface.operation.ModalContext$ModalContextThread.block(ModalContext.java:172)
	 at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:387)
	 at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:1002)
	 at org.eclipse.cdt.ui.wizards.CDTCommonProjectWizard.invokeRunnable(CDTCommonProjectWizard.java:177)
	 at org.eclipse.cdt.ui.wizards.CDTCommonProjectWizard.getProject(CDTCommonProjectWizard.java:155)
	 at org.eclipse.cdt.ui.wizards.CDTCommonProjectWizard.performFinish(CDTCommonProjectWizard.java:193)
	 at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:799)
	 at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:429)
	 at org.eclipse.jface.dialogs.Dialog$2.widgetSelected(Dialog.java:619)
	 at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:248)
	 at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	 at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4362)
	 at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1113)
	 at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4180)
	 at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3769)
	 at org.eclipse.jface.window.Window.runEventLoop(Window.java:827)
	 at org.eclipse.jface.window.Window.open(Window.java:803)
	 at org.eclipse.ui.internal.actions.NewWizardShortcutAction.run(NewWizardShortcutAction.java:136)
	 at org.eclipse.jface.action.Action.runWithEvent(Action.java:473)
	 at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:595)
	 at org.eclipse.jface.action.ActionContributionItem.access$2(ActionContributionItem.java:511)
	 at org.eclipse.jface.action.ActionContributionItem$5.handleEvent(ActionContributionItem.java:420)
	 at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	 at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4362)
	 at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1113)
	 at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4180)
	 at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3769)
	 at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1127)
	 at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	 at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1018)
	 at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	 at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:654)
	 at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	 at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:598)
	 at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	 at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:139)
	 at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	 at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	 at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	 at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:380)
	 at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:235)
	 at sun.reflect.NativeMethodAccessorImpl.invoke0(null:-2)
	 at sun.reflect.NativeMethodAccessorImpl.invoke(null:-1)
	 at sun.reflect.DelegatingMethodAccessorImpl.invoke(null:-1)
	 at java.lang.reflect.Method.invoke(null:-1)
	 at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	 at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
	 at org.eclipse.equinox.launcher.Main.run(Main.java:1515)

------
REPORT
------
anonymousId         e6d43d54-b045-4bfd-8e48-e252724f8de8
name                
email               
comment             
eclipseBuildId      4.5.1.M20150904-0015
eclipseProduct      org.eclipse.epp.package.cpp.product
javaRuntimeVersion  1.8.0_73-b02
osgiWs              win32
osgiOs              Windows10
osgiOsVersion       10.0.0
osgiArch            x86_64
logMessage          false
ignoreSimilar       false

-------
BUNDLES
-------
name                ilg.gnuarmeclipse.templates.core
version             2.5.3.201602091039

name                org.eclipse.cdt.core
version             5.11.0.201509131935

name                org.eclipse.cdt
version             8.8.0.201509131935

name                org.eclipse.cdt.managedbuilder.ui
version             8.2.2.201509131935

name                org.eclipse.cdt.ui
version             5.11.0.201509131935

name                org.eclipse.core.databinding.observable
version             1.5.0.v20150422-0725

name                org.eclipse.core.databinding
version             1.5.0.v20150422-0725

name                org.eclipse.core.resources
version             3.10.1.v20150725-1910

name                org.eclipse.core.runtime
version             3.11.1.v20150903-1804

name                org.eclipse.e4.ui.workbench
version             1.3.0.v20150531-1948

name                org.eclipse.e4.ui.workbench.swt
version             0.13.0.v20150504-0621

name                org.eclipse.equinox.app
version             1.3.300.v20150423-1356

name                org.eclipse.equinox.launcher
version             1.3.100.v20150511-1540

name                org.eclipse.jface
version             3.11.0.v20150602-1400

name                org.eclipse.swt
version             3.104.1.v20150825-0743

name                org.eclipse.ui
version             3.107.0.v20150507-1945

name                org.eclipse.ui.ide.application
version             1.1.0.v20150422-0725

name                org.eclipse.ui.ide
version             3.11.0.v20150825-2158
```

ilg-ul, 2016-02-10:
I confirm the problem, a fix will be available soon.

ilg-ul, 2016-02-10:
on updates-test is a new release. could you give it a try?

ilg-ul, 2016-02-10:
fixed in 2.11.3-201602101653

closed 2016-02-10 by ilg-ul

in4lio, 2016-02-11:
Yes, the problem is vanished) Thank you for so quick response.