eclipse-embed-cdt/eclipse-plugins #66: Installation of the second package is not executed and no message displayed

marek-trmac, 2016-02-11:
•plug-in version: feature GNU ARM C/C++ Packs (Experimental)	2.2.2.201602101653
•Java version: 8
•operating system: Win7 
•toolchain version: GNU ARM latest
•the detailed and complete list of steps to reproduce the bug
Open "Packs" perspective
Open "Packs" view
Right click on any uninstalled package and select "Install"
Immediatelly Right click on ANOTHER uninstalled package and select "Install"
Observation: only first package is being installed
Expectation: Both packages installed (or at least any error message displayed that second package is not installed because first package still being installed)

ilg-ul, 2016-02-11:
> Immediatelly Right click on ANOTHER uninstalled package and select "Install"

hmmm.... it was not intended to be used like this, to install multiple packages you can select multiple select them and click Install; in the mean time you can do anything you want, except install other packages.

the Packs Manager will suffer a major rewrite soon, so I'll probably change this behaviour too.

marek-trmac, 2016-02-11:
Hi. Sorry, this is not an enhancement, it is a bug. I just used command in the UI and it did not work. The installation of one package is quite long, you cannot consider it will be installed before user selects another package.

ilg-ul, 2016-02-11:
would you prefer a modal progress dialog while the packages are installed?

marek-trmac, 2016-02-11:
Definitely no modal dialogs

ilg-ul, 2016-02-11:
then what would be the expected behaviour? how about queuing all requests?

marek-trmac, 2016-02-12:
Queueing requests would be great!!!