eclipse-embed-cdt/eclipse-plugins #67: Showing OpenOCD and PEMicro under MacOSx when trying to "Install New Software"

mzan, 2016-02-12:
Hello,
we have an enhancement request for your OpenOCD and PEMicro features. The problem is, they are not supported on MacOSx, but we have them on our KDS updates site for Linux and Win, so when you try the "Install New Software" they show up and can be installed. Could you please add OS dependency on Linux and Win under the "Supported Environments" in those feature? It would prevent users from accidentally installing them. I guess this is reproducible on any Eclipse/Java/Mac/plug-ins versions. Thank you. 

Best regards,
Michal Zan

ilg-ul, 2016-02-12:
you must be missing something, GNU ARM Eclipse does not provide support for PEMicro, and OpenOCD was always available for OS X.

mzan, 2016-02-15:
yeah, you are right, sorry, my fault

closed 2016-02-15 by ilg-ul