eclipse-embed-cdt/eclipse-plugins #68: CMSIS-PACK Manager: Illegal http file size -1

ErichStyger, 2016-02-13:
GNU ARM C/C++ Packs (2.2.2.20160201653)

Packs manager works fine with the ARM.com/packs site.
I try to setup my own packs server, and the packs are verified and working fine with the ARM package manager, but I'm not able to get packs working with the GNU ARM packs manager.

I'm using the following packs repository:
Type: CMSIS Pack
URL: http://www.steinerberg.com/pack/index.idx

That site has the following files:
http://www.steinerberg.com/pack/index.idx
http://www.steinerberg.com/pack/McuOnEclipse.Utility.pdsc
http://www.steinerberg.com/pack/McuOnEclipse.Utility.1.0.0.pack

But when I connect to that site I get a "Read Error" in Eclipse
[image]
:
That file exists, but somehow the plugin is not able to read it the size of it?

Log (console):
2016-02-13 14:28:50
Update packs job started.
Parsing "http://www.steinerberg.com/pack/index.idx"...
Contributed 1 pack(s).
Missing "C:\nxp\CMSIS-PACKS\.cache\McuOnEclipse.Utility.1.0.0.pdsc", ignored by user request.
File "C:\nxp\CMSIS-PACKS\.cache\.content_www_steinerberg_com_pack.xml" written.
Update packs completed in 2s.

2016-02-13 14:28:53
Loading repos summaries...
Parsing cached content file "C:\nxp\CMSIS-PACKS\.cache\.content_www_steinerberg_com_pack.xml"...
File parsed in 1ms.
Identifying installed packages...
Found no installed packages.
Completed in 1ms.

2016-02-13 14:28:53
Collecting devices...
Found none.
Completed in 1ms.

2016-02-13 14:28:53
Collecting packs...
Found none.
Completed in 1ms.

Thanks,
Erich

ilg-ul, 2016-02-13:
I'll investigate.

in4lio, 2016-02-13:
I'm not sure, but have you tried to add slash to the end of url `<url>http://www.mcuoneclipse.com/pack</url>` into `.pdsc`, may be plugin cannot to find pack?

ilg-ul, 2016-02-13:
@in4lio, the url displayed in the error dialog seems ok, and trying to get it via curl was successful, so there must be something else that fails when trying to get the file size.

ErichStyger, 2016-02-13:
Yes, I have tried the addition slash, but it did not help. I can retry. I was thinking that somehow the server might block the access? But I thought it is a normal http access, as the file is accessible with the web browser.

in4lio, 2016-02-13:
Try to upload it on github like [here](https://github.com/in4lio/mdr1986x-pack-repo).

ErichStyger, 2016-02-13:
Interesting, it works for me with that GitHub. Maybe I should put things on GitHub too? But usually I'm of the opinion that sources files should be on GitHub, but not the packages (zip files). I saw that Liviu has one (dummy) pack repository on SourceForge. I have not tried SourceForge yet, does that work?

Thanks!

ilg-ul, 2016-02-13:
please keep the steinberg.com site up for a while, to have a chance to diagnose the problem.

> sources files should be on GitHub, but not the packages (zip files)

GitHub provides a great experience, everything is Git and everything is markdown. 

you can publish binary files in the project git, but even more appropriate is to use the project web (or GitHub Pages, as it is named). it is also stored as Git (a distinct branch or a separate repo), so the entire project is somehow limited (1 GB, if I remember right). the Pages site is available from a separate server, with a distinct url, ending in `github.io`.

on the other side SourceForge provides a distinct download solution, somehow more appropriate for high volume downloads, with multiple mirrors in different location. however, if you need a nice url, you also need to configure some web pages in the project web, and make them redirect pages (this is how I did it with the GNU ARM Eclipse update site).

> does it work?

well, SourceForge "generally" works, but it had some problems, sometimes the mirror servers take some time to update, sometimes the entire SourceForge is in read-only mode, etc.

personally I would like to get rid of SourceForge, and I think I know how to migrate everything to GitHub, (which, up to now, performed impeccably), but GitHub has a major disadvantage, it does not provide download statistics, which, for my needs, is very annoying.

ilg-ul, 2016-02-13:
> I thought it is a normal http access

it should be, but, in order to get the file size, it requires the server to provide complete http headers, otherwise it fails.

ErichStyger, 2016-02-13:
Yes, I will keep the site as is for diagnostics. Maybe my provider for whatever reason fails with providing complete http headers.

I'm trying now things with a GitHub site (https://github.com/ErichStyger/McuOnEclipse_CMSIS-PACKS).
I tried as well the site on https://github.com/in4lio/mdr1986x-pack-repo. 
While there is no error, no packs are shown in the Eclipse UI (for both sites).
Is this normal? I don't think so?

Thanks,
Erich

ErichStyger, 2016-02-13:
Oh, I take that back: after one more refresh the Milandr packs are showing up, sorry for the false alarm :(

ErichStyger, 2016-02-13:
BTW, ARM is using Schema version 1.4 in their tutorials, but it seems only 1.3 is accepted:
Processing "C:/nxp/CMSIS-PACKS/.cache/McuOnEclipse.Utility.1.0.0.pdsc" for content.xml ...
**Schema version "1.4" not recognised, package McuOnEclipse.Utility.pdsc ignored.**
File "C:\nxp\CMSIS-PACKS\__github_com_ErichStyger_McuOnEclipse_CMSIS-PACKS_raw_master.xml" written.
Processing "C:/nxp/CMSIS-PACKS/.cache/Milandr.MDR1986BExx.1.4.1.pdsc" for content.xml ...
**Schema version "1.3".**

ilg-ul, 2016-02-13:
> Schema version "1.4" not recognised

ah, please enter a new feature request.

ilg-ul, 2016-02-13:
the steinberg.com does not provide the `Content-Length:` header line, and the plug-in depends on it to get the file size:

```
HTTP/1.1 200 OK
Server: Apache
Last-Modified: Sat, 13 Feb 2016 13:04:08 GMT
ETag: "2ee837a-2c26-52ba66b4e9618"
Accept-Ranges: bytes
NZSpeedy: ON,O1
Content-Type: text/plain
Transfer-Encoding: chunked
Date: Sat, 13 Feb 2016 17:55:51 GMT
Age: 0
Connection: keep-alive
Via: 1.1 NetZone.ch-Speedy
X-Cache: MISS

<?xml version="1.0" encoding="utf-8"?>
...
```

probably the update logic should be immune to this.

ilg-ul, 2016-02-14:
> working fine with the ARM package manager

which ARM packs manager you are referring, the one in MDK?

ErichStyger, 2016-02-14:
Yes, the one in the Keil/ARM MDK.

ilg-ul, 2016-02-14:
> the one in the Keil/ARM MDK

how did you configure it to point to your server?

ErichStyger, 2016-02-15:
>>how did you configure it to point to your server?
I did not attempt to have http://www.keil.com/pack/index.idx list my server (my understanding is that I would have to submit a request to ARM according to their web site and it takes usually 7 days or so). I'm not in a state that things are stable (yet), I might consider this in the future. So I'm not able to use my site with the ARM/Keil Package manager as I have not found a way to point to my server. That's why I love your package manager as it provides me that freedom :-)

ilg-ul, 2016-02-15:
> That's why I love your package manager as it provides me that freedom :-)

:-)
 
frankly, it is still an experimental plug-in, with it's fair share of problems, I hope you'll like it more when stable and mature.

ilg-ul, 2017-04-24:
I tried again with the latest beta 3.4.1 version and the problem seems to be solved.

closed 2017-04-24 by ilg-ul