eclipse-embed-cdt/eclipse-plugins #69: CMSIS-PACK: Support for Schema 1.4

ErichStyger, 2016-02-13:
ARM is using Schema version 1.4 in their tutorials, but it seems only 1.3 is accepted:
Processing "C:/nxp/CMSIS-PACKS/.cache/McuOnEclipse.Utility.1.0.0.pdsc" for content.xml ...
Schema version "1.4" not recognised, package McuOnEclipse.Utility.pdsc ignored.
File "C:\nxp\CMSIS-PACKS__github_com_ErichStyger_McuOnEclipse_CMSIS-PACKS_raw_master.xml" written.
Processing "C:/nxp/CMSIS-PACKS/.cache/Milandr.MDR1986BExx.1.4.1.pdsc" for content.xml ...
Schema version "1.3".

ilg-ul, 2016-02-13:
please try version 2.12.1 from updates-test, it should support schema version 1.4.*

ErichStyger, 2016-02-14:
Yes, I confirm it works with that version. Multumesc!

closed 2016-02-14 by ErichStyger