eclipse-embed-cdt/eclipse-plugins #70: CMSIS-PACK: ability to run the pack refresh in the background

ErichStyger, 2016-02-13:
Doing a refresh of the package manager (to check the packs) takes a very long time (about 10 minutes or so on my machine. During that time there is modal dialog, and I cannot run the update in the background, so I have to wait.

I wish I could do this update in the background (similar to the build process).

[image]

ilg-ul, 2016-02-13:
right. the entire Packs manager will get a refurbish, I'll probably add a button to move the job to background.

marek-trmac, 2016-02-15:
It would be good to show package name, even the package is not downloaded yet. The details could be downloaded after the user clicks on the package.

ilg-ul, 2016-02-15:
> show package name, even the package is not downloaded yet

isn't it?

@marek-trmac, can you be more specific?

marek-trmac, 2016-02-15:
By default, Packs view is empty (see atatched screenshot).
[image]
After the user press "Refresh" button, all packges are downloaded (this is long process), however package names are listed in http://www.keil.com/pack/index.idx and could be displayed immediatelly.

marek-trmac, 2016-02-15:
To avoid this long process:
[image]

ilg-ul, 2016-02-15:
well, I considered this option when I wrote the plug-in, but it requires the plug-in to access all repository sites and download the index files during the Packs plug-in startup. what if access to one of these servers is slow and the entire step takes a few minutes? no, I think this kind of 'behind the scenes' actions may lead to poor user experience and should be avoided, I prefer to allow the user full control when to start these actions.

marek-trmac, 2016-02-15:
I agree to preserve user control over "Refresh" functionality. This can work together with automatic refresh on background, which would be much better user experience.

ilg-ul, 2016-02-15:
@marek-trmac, please open a new enhancement request, specifically for the initial state of the packs manager window, with explicit request that the process should happen in the background.

cmorty, 2019-06-28:
As this is quite old and it seems like work has stalled. How much trouble would it be to make this non-modal (Add a send to background button). That way one could at least continue working while updating.

ilg-ul, 2019-06-28:
My interest in CMSIS-PACKS is getting lower and lower.

I'll consider this, but I can't promise anything.

Do you need the packs for anything else than using the SVD during debugging?

cmorty, 2019-06-28:
@ilg-ul No worries. I only need them for debugging and while waiting for the packs to update, which I do rather seldom, I searched the bug-tracker and found this rather old bug. :)

ilg-ul, 2019-06-28:
For debugging only you can bring the SVD file to your project and point the debugger to it.

Actually I was thinking to make this use case the default, and avoid the hurdle with CMSIS Packs at all.

Could you try if this acceptable?

cmorty, 2019-06-28:
@ilg-ul Using the packs is super-convenient as it allows you to select the device via the project settings dialog. Maybe having a script that downloads them, repacks the relevant data and provides that via github.io would be a convenient solution. I haven't looked into licensing issues, though.

ilg-ul, 2019-06-28:
I know, but the way CMSIS Packs are organised, not to mention that the index is most of the time out-of-sync with actual packages, requires very creative solutions to manage them (read 'complex logic'), and I don't have time now for this. :-(

ilg-ul, 2019-06-28:
Actually, for ST devices, I'm considering moving much closer to CubeMX.

Do you know if the packages provided now by CubeMX include the SVD files?

cmorty, 2019-06-28:
@ilg-ul  Just checked my CubeMx repo folder: There are no SVDs and the folder structure does not look like that's likely to happen (might be they are hidden in some zip, but I don't think so).

ilg-ul, 2019-06-28:
too bad...

:-(

ErichStyger, 2022-01-25:
closing old issue: KDS is legacy now.

closed 2022-01-25 by ErichStyger