eclipse-embed-cdt/eclipse-plugins #72: CMSIS-Pack Plugin: KLxx Series installed but it does not show up?

ErichStyger, 2016-02-14:
I have the Kinetis_KLxx_DFP installed, and the Keil/ARM CMSIS Device viewer shows the device: however it is not shown in the 'Devices' list (not black):
[image]

The problem is that with that somehow the KLxx devices do not show up in the Project settings, not sure why?
[image]

ilg-ul, 2016-02-14:
this looks like a small bug. if you exit/enter Eclipse, does it show it properly?

otherwise, for questions please use the project forum https://www.element14.com/community/groups/gnu-arm-eclipse

ErichStyger, 2016-02-17:
enter/exit Eclipse does not change things. What is the condition that the icon under the Device view turns from 'gray' to 'black'?

ilg-ul, 2016-02-17:
then it is definitely a bug, the mere presence in the installed packages should be enough.

ErichStyger, 2016-02-20:
closing issue, will file a bug report

ilg-ul, 2016-02-20:
No need to, I already marked it as a bug, don't you see this label?

ilg-ul, 2016-02-20:
duplicate of #85.

ErichStyger, 2022-01-25:
closing old issues

closed 2022-01-25 by ErichStyger