eclipse-embed-cdt/eclipse-plugins #82: pyOCD Debugger launch config tab shouldn't populate targets combobox with static list

sukidog, 2016-02-19:
[fix on the way]
The implementation invokes the pyOCD gdbserver to determine what boards and processors ("targets") it supports and makes those choices available in the Debugger tab. However, if first populates the processor combobox with a static default list. That's misleading and ultimately can do more harm than good.

ilg-ul, 2016-02-19:
For the pyOCD plug-in (currently not included in the public release), please coordinate with Chris @flit, he added it and he will maintain it.

sukidog, 2016-02-19:
We work together. (He is about 100' down the hall from me). He's given me write access to his repository. I'll be submitting fixes, but first creating bugs for them. Expect most pyOCD plug-in activity to come from me.

ilg-ul, 2016-02-19:
Great, then both of you will maintain the plug-in! :-)

ilg-ul, 2017-04-24:
was this issue solved?

flit, 2017-12-01:
@ilg-ul This issue has been fixed.

ilg-ul, 2017-12-01:
great, please submit the pull requests.

flit, 2017-12-13:
Sorry, I meant that the fix has already been merged and you can close this issue. (Same with #83.)

closed 2017-12-13 by ilg-ul