eclipse-embed-cdt/eclipse-plugins #83: pyOCD launch config tab has context issues

sukidog, 2016-02-19:
The implementation mismanages the active launch config context. E.g., if you have two launch configurations that each point to a different pyOCD gdbserver, and each of these reports different targets and boards, the same, single collection of targets and boards can end up being shown in both launch configurations. This happens when you have the launch configs open to the Debugger tab and you switch back and forth between the two launch configs. In that case, ILaunchConfigurationTab.activated() is not called by Eclipse as you go back and forth, and so the current implementation ends up using a single context (which is correct only half the time).

ilg-ul, 2016-02-19:
When you add issues, please label them properly and assign them to the person expected to fix them.

sukidog, 2016-02-19:
I see no UI for labeling and assignment.

ilg-ul, 2016-02-19:
do you see the Labels, Milestone, Assignee fields in the right of the issues page?

sukidog, 2016-02-19:
Yes, but I see no UI for setting those fields when I create the ticket, nor any way to set those fields after ticket creation. I'll email you a screenshot of the create-issue UI I see.

ilg-ul, 2017-04-24:
was this issue solved?

flit, 2017-12-01:
@ilg-ul Yes, this issue was resolved.

closed 2017-12-13 by ilg-ul